hr-js/sta_access_manager #10: Raspberry Piのipアドレス固定しつつ、Wifi接続する

kakaka19, 2017-09-30:
Raspberry Piはwifiに接続するたび、異なるipアドレスが割り振られる。その場合個人環境からssh接続を行う際に、毎回ipアドレスを確認しなければならない。

そのため、ipアドレスを固定する必要があるが、ipアドレスを固定すると、社内wifiから外部(githubなど)へ出れない(確かwifi自体には繋がってたはず・・・忘れました)。

Raspberry Piの環境構築はほぼ完了しているが、作成したモジュールなどのデプロイはgitで行うことを想定しているので、githubの参照は必須。

- 試行案

以下のHPを見つけた。これで解決するか？
[Raspberry Piでいろいろなネット環境でつないで外部にだけつながらない時の対処法](http://www.1ft-seabass.jp/memo/2015/05/15/raspberry-pi-dhclient-refresh-wlan/)

- 参考

ipアドレス固定手順
[Raspberry Pi 3 (Raspbian Jessie)の無線LANに固定IPアドレスを設定する#固定IPアドレス設定](https://qiita.com/momotaro98/items/fa94c0ed6e9e727fe15e#%E5%9B%BA%E5%AE%9Aip%E3%82%A2%E3%83%89%E3%83%AC%E3%82%B9%E8%A8%AD%E5%AE%9A)

ikkyu-3, 2017-10-01:
> 試行案
これで解決したとあるけど、根本的な原因解決はできていないように思えます。
仕様なのか、バグなのか。。。なんか気になってしまう。

とりあえず接続できればいいのだけども。。

ikkyu-3, 2017-11-04:
今回対応しないことになったので、Issueはcloseします。

closed 2017-11-04 by ikkyu-3