hr-js/sta_access_manager #12: githubの機能を活用する

kakaka19, 2017-10-14:
現時点で使用しているgithubの機能は以下の通り
- Issue
- Wiki
- Pull requests

他にもいろいろなことができるはずなので、最大限に活用したい。思いつくものとしては
- Label
- Projects
- Milestone
- Issue Template

ただ、活用することだけに着目して開発フローが滞ると意味がないので、全体の共有・意見収集は必須。

ikkyu-3, 2017-10-15:
[こっそり始めるGit／GitHub超入門（12）：開発者のスケジュール管理に超便利、GitHub Issues、Label、Milestone、Projects使いこなし術 (1/4) - ＠IT](http://www.atmarkit.co.jp/ait/articles/1704/28/news032.html)

このサイトが参考になりそうな気がします。
時間見つけて目を通しておくといいかも

ikkyu-3, 2018-01-14:
@kakaka19  
運用に関してだけど、閲覧の設定ができる...と聞いて調べて見たけどよくわからなかったから、URL貼ってもらっていいですか？

kakaka19, 2018-01-15:
@ikkyu-3 
slackでも送りましたが、wikiとかソースは基本publicみたいです。（勘違いしていたようです。すみません。。。）
調べて見たんですが、ドキュメントの共有サービスは有料が多いみたいなので、Google Driveとslackの連携を調べてみます

ikkyu-3, 2018-01-16:
@kakaka19  
調査ありがとう！

> ドキュメントの共有サービスは有料が多いみたいなので、Google
 Driveとslackの連携ができないか調べてみます。

対応ありがとう！確認しました！どうなるのかちょっと試してみたいねー

> wikiだけ、bitbucketにお引越しとかも...ありすかね。

ありだと思う。無料枠(人数制限的な意味で)の確認してみる必要があるかも、、これは...とりあえず、無料枠に関してはこっちで調査しときます。その後、現状いけそうなら検討してみるって感じでどうでしょう？

ikkyu-3, 2018-01-16:
5ユーザまで無料だね
https://ja.atlassian.com/software/bitbucket/pricing?tab=cloud

んー、やり方考えればいけそうだね。
有料枠という選択肢、一度交渉してみるってのもいいかもって思ってはいるかなー

kakaka19, 2018-01-16:
@ikkyu-3 
http://japan.blogs.atlassian.com/2012/09/refer-a-friend-to-bitbucket-for-free-users/

bitbucketのアカウントを持っていない人ならさらに+3人まで招待できるみたいですね。ただ2012年の記事なので、今でも可能なのか怪しいですが..

一旦原点に立ち戻って考えたのですが。
__Google Driveのドキュメントが更新されても、他のメンバーが更新を知ることができない__
っていうのが最大のネックだと思っています。

なので、以下のフローに沿ってやれば、現状、GitHubのissue + Google Driveで解決できると思っています。不便は不便ですが...
1. githubに仕様関連のissueを作成する
1. Google Driveのドキュメントを更新する
1. issueをクローズする

ドキュメント運用に関しては、切りわけて考えた方がいいのかもしれません。

kakaka19, 2018-01-27:
適宜使うことを検討するのでクローズ

closed 2018-01-27 by kakaka19