hr-js/sta_access_manager #15: 開発フローを検討する

kakaka19, 2017-10-14:
#14 を行うにあたって、開発フローを決定したい。

現時点(2017/10/14)で決まっている事項は以下の３つ。全体への共有も含めて記載しておく
- developにコミットはしない
- branchはgithubのアカウント名で作成する。
- 開発用branchからPull requests→他の人がレビュー→OKならdevelopにmerge

最低限、上記の項目を踏まえていれば開発は可能。

検討事項としては以下が挙げられる。
- gitで管理しないファイルの検討(個人情報等)
- 仕様決定→設計→実装といった開発における大まかなサイクルの決定
- レビュー観点

ikkyu-3, 2017-10-15:
> gitで管理しないファイルの検討(個人情報等)

Slackでファイル共有できるので、個人情報などの情報に関しては、こちらで共有するようにするのはどうでしょう？

kakaka19, 2017-10-15:
個人情報に関してはSlackでいいですね。ただサーバのIPアドレスとか、テスト用のICカードのID?とか、微妙なラインが気になりますね。漏出することよりも、この情報はpushしてもいいのか？って迷わないようにあらかじめ決めておきたいです。

>レビュー観点

レビューに関しては、Airbnbのコーディング規約が参考になりますかね。
[Airbnb JavaScript スタイルガイド](https://mitsuruog.github.io/javascript-style-guide/)

ikkyu-3, 2017-10-15:
> ただサーバのIPアドレスとか、テスト用のICカードのID?とか、微妙なラインが気になりますね。漏出することよりも、この情報はpushしてもいいのか？って迷わないようにあらかじめ決めておきたいです。

その情報は、アプリが持っている情報なのかで判断するのはどう？
例えば、
`ICカードのIDは、カードが持っている情報だからpushしない。`  

わかりやすいでしょうか？

kakaka19, 2017-10-15:
なるほど、それはわかりやすいですね。それなら、共有しない情報としては今の所
- サーバのIPアドレス、ホスト名
   - 各々設定ファイル(.yml?)を作成することで回避。設定ファイルはpushしない
- DBサーバのURL
- ICカードのID

ぐらいですかね。

ikkyu-3, 2017-10-16:
だね。それでいいと思います！

ikkyu-3, 2017-10-17:
1. 開発フローのドラフト作成
1. 開発メンバーはハンズオンで一度一通りの作業をやってみる
1. 反省会
1. 開発フロー作成
1. READMEとWIKI作成

こんな感じでどうでしょう？
勉強にもなるし、課題など新たな発見があるかと思います。

kakaka19, 2018-01-27:
都度issueにして検討する

closed 2018-01-27 by kakaka19