hr-js/sta_access_manager #27: フロント画面開発の仕方に関して

ikkyu-3, 2017-12-03:
画面のレイアウトが出揃いました。
そこで画面開発に関して一度、意見がほしいです。

## 現在の開発フロー
1.  待ち受け画面をtankette148、Tchikawa、toyoshimmer がHTMLで作成する
1.  作成した3つの中から、1つをアプリに使用するHTMLとする
1. ikkyu-3、minako-nakanishiがReactで使用するためjsxに変換する
## 問題点
- 他の画面の開発の進捗が進まない
- 頑張って作成したHTMLが使用されないのはやっぱり悲しい
## 提案
1. 待ち受け画面は現状のまま３人が作成します(変更なし)
1. 他の画面はそれぞれ担当を決める(サポートはikkyu-3)
1. 出来たらminako-nakanishi(スキル的にtoyoshimmerもできるかと思います)にはPresentational componentで使用するためjsxファイルを作成してもらう(サポートはikkyu-3)
## 考慮する点
- SVGはどうする？みんなで作るならサイズだけ決めておきたい

以上です。どうでしょう？

minako-nakanishi, 2017-12-03:
Presentational componentは不安なので、toyoshimmerにも参加していただきたいです。

Tchikawa, 2017-12-04:
画面担当決めたいっす！

toyoshimmer, 2017-12-04:
Presentational componentやります！

ikkyu-3, 2017-12-04:
@toyoshimmer
画面開発とPresentational component対応の両方やるってことでいいですか？

toyoshimmer, 2017-12-04:
そうです！

tankette148, 2017-12-04:
問題点、1はある程度は勉強と割り切る、
2は担当決めれば進められるかと！
なので提案そのまま進める感じで。
とりあえず担当画面決めましょー！

SVGは画面サイズ的に280px*400pxくらいで良さそうかと思いました。

ikkyu-3, 2017-12-04:
@みなさん
了解です。次のMTで担当きめましょう！

> SVGは画面サイズ的に280px*400pxくらいで良さそうかと思いました。

一度そのサイズで見え方をみんなで確認したいです。

kakaka19, 2017-12-04:
お疲れ様です！
横槍で申し訳ないのですが。管理者画面の方も併せて話し合ってくれるとサーバーサイドとしては助かります！手が止まってしまうので。
何となくで良いので決めてほしい事項としては
- グラフを出したいか、D3.jsを使うのか
- 訪問者を一覧として出したいか(これは結構重要です)
あと、仕様として管理者側がどのような機能を欲しているのかは気になります。
現時点だとラズパイ側のUI(UX?)がメインですが、併せて決めましょう。というか、管理者側のUi、もしくは必要条件が揃わなければ、API側の仕様が決定できません。
@Tchikawa @ikkyu-3 @tankette148 の負担が大きくなるやもですが、とりあえず変更が前提としたモックでもいいので、決めてしまいましょう。これマジな。

tankette148, 2017-12-04:
画像フォルダ直下に280px*400pxと250px*450px(こっちのがいいなと思って調整したやつ)おいときました。
微調整は話し合いながらのが良さそうかと

ikkyu-3, 2017-12-06:
話し合い完了したのでクローズ

closed 2017-12-06 by ikkyu-3