hr-js/sta_access_manager #31: キー入力画面(モック)の作成

ikkyu-3, 2017-12-06:
共有したキー入力画面(モック)の作成をお願いします。
モックなので、htmlとcssで作成してみてください。

以下のことを考えて実装してもらえるといいかと思います。
- inputタグを使用しない方法で実装
- 足りないキーを作成

対応が遅くなるかもしれなですが、サポートしますのでよろしくお願いします。

ikkyu-3, 2018-01-28:
## 目標
番号を押したらテキストボックスに番号を表示するところまでで大丈夫です!

minako-nakanishi, 2018-01-28:
承りました！

minako-nakanishi, 2018-01-28:
番号ボタンの配置でtableタグを使おうかと思っているのですが、tableタグっであまり使わない方が良いんでしたっけ？
Pタグでも良いかなと思ってはいるのですが。。

ikkyu-3, 2018-01-29:
> 番号ボタンの配置でtableタグを使おうかと思っているのですが、tableタグっであまり使わない方が良いんでしたっけ？
Pタグでも良いかなと思ってはいるのですが。。

pかtableかというと、pは使わないですね。↓ここみると、使用しないかなと、、、
[p 要素 - HTML | MDN](https://developer.mozilla.org/ja/docs/Web/HTML/Element/p)

グリッドレイアウトを使うときれいに配置できると思います。
[CSS グリッドレイアウト - CSS | MDN](https://developer.mozilla.org/ja/docs/Web/CSS/CSS_Grid_Layout)

グリッドレイアウトって検索するとわかりやすく説明してくれてるサイトいっぱいあるので、
一度検索してみてください。

minako-nakanishi, 2018-01-29:
ありがとうございます！

ikkyu-3, 2018-01-30:
[ユーザのマスターデータに関して](https://github.com/hr-js/sta_access_manager/issues/43)
案2を採用することになったので、こちらのモックは作成しないことに変更します。

よってcloseします。

closed 2018-01-30 by ikkyu-3