hr-js/sta_access_manager #35: ドキュメントの運用を検討する

kakaka19, 2018-01-16:
## 目的
ドキュメント(仕様書、設計書)の運用を検討する。

## 背景
資料をGoogle Driveで管理しているが、以下の問題がある。別のサービスを利用することで、解決できないかを調査する。
- Google Drive上のドキュメントを更新しても、通知がいかない。
- ローカルで資料作成 → アップロードが手間
- 差分がわからない

## ゴール
Google Drive上の資料を精査し、公開して問題が発生するか判断する。

1. 発生しえない場合
運用ポリシーを決め、Githubのwikiで運用する。
1.  発生する可能性がある場合
閲覧制限が可能なドキュメント運用サービスを調査する。

2に関してはいくつか候補が上がっている。無料であることが前提。
- Bitbucket(wikiのみ)
- [Crowki](http://site.crowi.wiki/) + [Heroku](https://www.heroku.com/)
- Google DriveとSlackを連携

kakaka19, 2018-01-16:
Google Drive上の資料をファイルごとに精査

チェック項目
- 個人を特定できる社内情報が書かれていないか？
- 個人名/個人メールアドレスが書かれていないか？
- そもそも公開/共有する必要があるものか？

チェック対象ディレクトリ
- [ ] そのほか
- [ ] 画面
    - [ ] 画面原案
- [ ] 活動ログ関連
- [ ] 仕様・設計関連

### そのほか
画面側のアーキテクチャ。写真だが、図に整形する必要はある。公開しても問題なし

### 画面
画面側のモック(jpg)。公開しても問題なし。写真にニックネームが書かれているが、これは共有/公開する必要がない。

### 画面原案
モックの原案(写真)。図に整形する？画面のモック(jpg)ができているなら共有する必要がない気がする。

### 活動ログ関連
タスクを割り振ったテキストファイルが一つ存在。公開/共有する必要なし。githubのissueにする

### 仕様・設計関連
ファイルの内容が多岐なため、一つ一つ確認する。

#### api.txt
APIのI/O。社員番号例を別のものに置き換えて共有
####  sta_access_managerの機能.txt
システム全体の機能仕様を俯瞰したもの。公開しても問題はない
#### sta_access_managerの要求・要件.txt
システム要件を記載したもの。公開しても問題はない
#### ヒアリング内容.txt
クライアントへのヒアリングを記載したもの。公開しても問題はない。
#### 管理者画面案.txt
管理者側のアクティビティから画面仕様を記載したもの。公開しても問題はない。

krhrtky, 2018-01-16:
@kakaka19 @ikkyu-3 
概ね問題ないように思います。

> ヒアリング内容.txt

記載内容にちょっと具体的な記述があったので、そこを変えておけば堅いかなと。

ikkyu-3, 2018-01-16:
### その他
図にする必要ありでおkです！
### 画面
キャラクターに関して。自作したものか、公開しても大丈夫なのか確認したいです。その他は同意見。
### 画面草案
ラフ絵を清書したのが画面と判断でいきましょう。なので必要なしという意見です。
### 活動ログ関係
同意見
###  仕様・設計関連
#### api.txt
同じ認識。ユーザIDはどうでしょう？
#### sta_access_managerの機能.txt
名称を修正しておきたいです。登場人物の()内は公開する必要がないと判断はどうでしょう？
#### sta_access_managerの要求・要件.txt
同意見
#### 管理者画面案.txt
同意見。用語を他と統一する必要がありって思ってます。

すこし濁して書いていてよかった。。

ドキュメントの修正で公開できるでおkです！

Tchikawa, 2018-01-16:
**画面**
キャラクターに関して。拾ってきたやつに色つけたやつなので...アウトだと思います...すみません....

ikkyu-3, 2018-01-16:
@Tchikawa 
ありがとう！サイトに明示的に使用方法など書いてない場合は、使わないという判断は正しいです。

@kakaka19 @krhrtky @Tchikawa
何か画像を作って、差し替えで対応でどうでしょう？

Tchikawa, 2018-01-18:
画像差し替え今週中に対応します

kakaka19, 2018-01-19:
ドキュメント運用検討

以下の要項を判断基準としました。優先度順で記述しています

1. 編集が楽かどうか
1. ドキュメント更新の通知が行くかどうか
1. 閲覧制限が可能かどうか
1. 情報が探しやすいかどうか

### 案1

Github wiki

__positive__
- setttingsから`Restrict editing to collaborators only `のチェックを外すことで`メンバーのみに閲覧を制限`できる(できました！)。
- 自動でページ一覧を作成してくれる
- ローカルでなら`git log`で差分を確認できる

__negative__
-  __Slackに通知できない__

### 案2

Github Pages

__positive__
- ソースリポジトリで管理するため、__更新のたびにSlackに通知できる__
- ソースリポジトリで管理するため、差分がブラウザで確認できる
- ドキュメント更新でプルリクエストが作成可能。ドキュメントのレビューができる

__negative__
- 更新が失敗した際、メールで通知されるが、__エラーメッセージが雑__
- pushするとページが更新されるが、なかなか反映されない
- 編集方法を共有する必要がある
- 閲覧制限ができない
- 情報がどこにあるかわかりにくい。目次を作る必要あり。

個人的には通知は諦めて、githubのwikiで運用してもいいかなと。というか、メンバー全体に通知する必要ってあんまりないかな...とか考え始めました。

krhrtky, 2018-01-20:
@kakaka19 
 調査ありがとう！
↑の内容を受けて、もう一つ候補の提案。

### gitでmarkdownファイルの管理をする

- イメージとしては、案１・２の中間

**positive**

- 編集が楽。markdown記法を知っていれば書ける(案1)

- ローカルでならgit logで差分を確認できる(案1)

- ソースリポジトリで管理するため、更新のたびにSlackに通知できる(案2)

- ソースリポジトリで管理するため、差分がブラウザで確認できる(案2)

- ドキュメント更新でプルリクエストが作成可能。ドキュメントのレビューができる(案2)

- JSの勉強会のまとめをしているのが、参考になるかと

   https://github.com/hr-js/learn_js_function/blob/master/doc/minutes_1.md

**negative**

- ページ一覧はない

    Chromeの拡張機能を使えば、ある程度解消できる(Chromeに限るのはなんとも言えない)

    https://medium.com/@osamingo/amazing-three-extensions-for-github-b01bac834040

- 閲覧制限ができない

あくまで参考意見としてという部分もありますが、こんなところでいかがでしょう。

kakaka19, 2018-01-20:
@krhrtky 

ソースと一緒に管理するっていうのはあり。でもたまに変に文字化け？するっぽいんだよね...

ikkyu-3, 2018-01-20:
@kakaka19 @krhrtky 
調査や提案ありがとう。

_markdownファイルを管理する_ 
これいいな！って思いました。

> メンバー全体に通知する必要ってあんまりないかな...とか考え始めました。

確かに。。

> たまに変に文字化け？するっぽいんだよね...

[Vim - VSCodeで編集したmarkdownをgithubで表示すると文字化けする(85390)｜teratail](https://teratail.com/questions/85390)
使用するエディターっぽいですね。

githubのwikiをmarkdownファイルに変えて、目次を作った場合のリスクが高いとなーって思ってます。

> 個人的には通知は諦めて、githubのwikiで運用してもいいかなと。

これを採用するってことでどうでしょう？

kakaka19, 2018-01-21:
@ikkyu-3
@krhrtky

ひとまず、wikiで行きますか！やっぱり通知がほしいってなったらソースリポジトリに移行しましょう。

ikkyu-3, 2018-01-21:
それでいきましょう！

krhrtky, 2018-01-21:
@kakaka19 
いいと思います！