hr-js/sta_access_manager #36: 画面側のアーキテクチャの図を作成する

ikkyu-3, 2018-01-21:
[ドキュメントの運用を検討する #35](https://github.com/hr-js/sta_access_manager/issues/35)関連

** 現状
画面側のアーキテクチャは、ホワイトボードにかいた図を写真で撮ったもの
** 目的
図に清書し、wiki(現在追加する箇所で検討しているのは、Homeの構成部分)に追加する

画像を図にするだけですが、wikiへの追加箇所について相談できればと思います。
- Home(現在はこっちで考えています)
- 別ページ

ikkyu-3, 2018-01-21:
図ができました。

wikiの追加箇所に関して、認識合わせができたらアップします。

kakaka19, 2018-01-21:
@ikkyu-3 
利用者向けの画面遷移図でしょうか？

それでしたらHomeよりも、uiの方が適切かなと思います。

https://github.com/hr-js/sta_access_manager/wiki/ui

ikkyu-3, 2018-01-21:
@kakaka19 
> 利用者向けの画面遷移図でしょうか？

遷移図ではなく、Google Driveの入室管理システム / そのほかディレクトにある画像ファイルです。
説明が下手くそでした。すいません。

> 画面側のアーキテクチャ。写真だが、図に整形する必要はある。公開しても問題なし

とのことだったので、図にしました。

> それでしたらHomeよりも、uiの方が適切かなと思います。

ですね。uiに追加します。ありがとう

ikkyu-3, 2018-01-22:
@kakaka19 @krhrtky 
uiに追加しました。
https://github.com/hr-js/sta_access_manager/wiki/ui

closeするタイミングの相談なんですが、

1. 追加してclose
1. 誰かに確認してもらってからclose

どれがよさそうですかね？

krhrtky, 2018-01-22:
@ikkyu-3 
完了する基準が必要だと思うので、２で良いと思います。
２の確認する基準を明確にしておくと、行き違い・手戻りがなくなるんじゃないでしょうか。

### 例

1. 写真を図にしたものに抜け漏れがないか確認する

1. レビュアー(フロント実装者を想定)がアーキテクチャとライブラリの大まかに理解できる。

1. 共有している画像ファイルとWikiの内容の対応関係を把握する。

等

kakaka19, 2018-01-22:
@ikkyu-3 @krhrtky 

自分も2でいいと思います!

ドキュメントの運用/更新ポリシーも用意しなきゃ...

ikkyu-3, 2018-01-22:
> 行き違い・手戻り

これが発生するのは、避けたいです。

では、ドキュメントの運用/更新ポリシーを決めてから、closeします。

closed 2018-01-27 by ikkyu-3