hr-js/sta_access_manager #38: フロント側の実装スケジュールを検討

ikkyu-3, 2018-01-21:
色々(モック画面)など進んできたので、リリース予定を仮で3月中と決めてみました。
期限を決めた方がいいと判断したためです。

残りの課題として
- 各画面の機能定義したドキュメントを作成する。PDFで作成する予定です。
- 残りの画面のモック
- コンポーネントの作成
- reactとreduxの連携
- モジュールの作成
- 現在のIssuesの消化(調査など)
- テストケース作成

やることは多そうですが、自分が一度経験しているものが多いので、３月中と見通しをたてました。
進捗はとっていき、リスケする判断も必要になってくるとは思います。

何か意見あれば、よろしくお願いします。

krhrtky, 2018-01-22:
@ikkyu-3 
すみません。現状の担当があまり把握できてません。

あと、個人によってスタート時点と業務その他の兼ね合いだったり、
モチベーション等々に変化があると思うので、
必要であれば先に割り振りの見直しもしておいて方が、いいのかなと感じました。

ikkyu-3, 2018-01-22:
@krhrtky 
> 現状の担当があまり把握できてません。

たしかに。
振り直します。少し時間かかってしまうかもしれないがお待ちください。

ikkyu-3, 2018-01-23:
- 現在の担当の振り直し
- モック画面作成のissue作成
を行いました。
**Assignee**で絞ってもらえるとわかりやすいかと思います。

残りは、もう少しお待ちください。ほとんど自分のissueです、、、

kakaka19, 2018-01-24:
@ikkyu-3 
@krhrtky 

スケジュールに関してなのですが、フロントサイドは関わる開発メンバーが多いので、Milestoneで管理してはどうでしょう。

一定の期間(2週間ごと)のMilestoneを作成しましたので、Developmentに入っているissueに関しては、とりあえず1st iterationに全て移し、そこから開発メンバー各々で実装完了できる期間に移して貰うっていうのはどうでしょうか。slackでも周知しますが。

ikkyu-3, 2018-01-24:
> スケジュールに関してなのですが、フロントサイドは関わる開発メンバーが多いので、Milestoneで管理してはどうでしょう。

賛成です！提案ありがとう。一定期間ごとの方がいいですね。
ちょっと長いスパンだったなと思います。

kakaka19, 2018-01-27:
@ikkyu-3 

スパンが長いのは問題ないのですが、Milestoneを使うことで以下の課題が解決できると思っています。

- 誰がなんのissueを__いつまで__にやるのか明確になる。
- 期間内にissueが達成できた/できなかったが共有できる。
    - 達成できなかった場合、周りがサポートを申し出ることができる、もしくはリスケできる。

ikkyu-3, 2018-01-27:
Milestoneで使用しでいく方法で行きましょう！

ikkyu-3, 2018-02-05:
Milestoneで管理し運用がはじまったので、このissueをcloseします。

ご意見ありがとうございました。

closed 2018-02-05 by ikkyu-3