hr-js/sta_access_manager #39: APIのデータモデルの型に関して

ikkyu-3, 2018-01-21:
## 現状
記載したのは全て `""` がついているので、string型と判断できる。

作成したのは自分ですが、
true, falseは、boolean型にしたほうがいいのでは？と考えています。

NGのmessageは「０ or 1」(文字列)で定義しています。
文言変更を考慮してステータスコードの意味合いとしてと思いましたが、プロパティをステータスにすればいいのでは？
などなどです。

## 変更の影響
- 変更した際の影響範囲、開発の進捗に影響が出るなど知りたいです。
- フロント側はこれからなので、影響なし

## 目的
修正ありでもなしでも、wikiにデータの型を明示したいと思います。

どうでしょうか？
ご意見もらえると嬉しいです。よろしくお願いします。

kakaka19, 2018-01-22:
@ikkyu-3 
そもそも、みたいな感じになって恐縮なのですが....
レスポンスのbodyに処理の成否を入れるのは画面側での判断が行い辛くないでしょうか？

[ユーザ登録](https://github.com/hr-js/sta_access_manager/wiki/api#%E3%83%A6%E3%83%BC%E3%82%B6%E7%99%BB%E9%8C%B2)を例にとりますが、

statuscode:200で4つの分岐が発生します。若干憶測が混じってますが...
- result :true 
    - -> ICカード登録画面へ
- reslut:false
     - -> message:0
         - -> 不正なIDなのでエラー画面へ
     - -> message:1
         - -> 登録済みだが有効なIDなのでICカード登録画面へ？

以下のようにstatus codeで判断できるよう、レスポンスを変えた方がいいのかなと提案します。

- 200番台（正常系:登録済みでも関係ない）
    - -> ICカード登録処理
```json
{
    "name": "ユーザの名前"
}
```
仮に入力されたユーザIDが登録済みだとしても、画面側ではそれを意識しない方がいいのかなと。

- 500番台 (異常系:サーバ側処理エラー)
    - -> 登録失敗(Elasticsearchにコネクションを持てないなど。利用者に通知することなのか微妙ですが。)
```json
{
    "message": "ユーザの登録処理が失敗したことを示すメッセージ"
}
```

400番台に関しては#40 でフロントサイドで処理を実装するようなので、触れていないです。

krhrtky, 2018-01-22:
@ikkyu-3 
@kakaka19 のステータスコードで分岐の方向のほうがいいかと思います。

- `result` の意味を考える必要が減る。

- 正常に処理した場合でも失敗と判断するケース(上記、登録済み 等)もエラーであると考えたほうがわかりやすい。

[MDN status codes](https://developer.mozilla.org/ja/docs/Web/HTTP/Status) 参照

一方で厳密に↑の方法で定義できるかという部分もあります。

例えば、複数のエラーが有ることを表現したいときです。

その場合は、`result` のvalueをbooleanではなく、enumのようなもの(jsにenumがないため)を定義して、判定する必要があると思います。

どちらにせよ`message` に0/1が入るのは違和感かなと

ikkyu-3, 2018-01-22:
@kakaka19 @krhrtky 
意見ありがとう。業務の癖が・・・

> 処理の成否を入れるのは画面側での判断が行い辛くないでしょうか？

ifやswitchで条件分岐させるので、行い辛いって感じはないです。
ただ、bodyのデータ構造が深かったり、複雑であると辛いなって思います。

> status codeで判断できるよう、レスポンスを変えた方がいい

これいいですね。
200で登録(body=nameあり)
204で登録済み(responseのbodyなし)

不正なIDは400系ですかね。。

一応少し気になって調べて見ました。
[WebAPIでエラーをどう表現すべき？15のサービスを調査してみた - Qiita](https://qiita.com/suin/items/f7ac4de914e9f3f35884)

オブジェクトを返してるっぽい。詳細な情報が必要だからってことなんだろうと判断はできるので、
これを真似る必要はなさそうですね。

> どちらにせよmessage に0/1が入るのは違和感かなと

賛成です。無理に同じオブジェクトの構造を適用しようとしたからこうなってしまったので、
ここは確実に変更ですね。

ということで

**ステータスコードで分岐の方向**
にしようかと思いますが、どうでしょうか？

ちょっと定義がかわりますが、今よりスッキリしそうな気がしてます。
フロントも楽です。

kakaka19, 2018-01-22:
@ikkyu-3 
@krhrtky 

>MDN status codes 参照
一方で厳密に↑の方法で定義できるかという部分もあります。
例えば、複数のエラーが有ることを表現したいときです。
その場合は、result のvalueをbooleanではなく、enumのようなもの(jsにenumがないため)を定義して、判定する必要があると思います。

この辺、好みかもしれないんですが、複数のエラーが存在することをフロントサイドに通知する必要があるか疑問です。例えば500番台のエラーを利用者/管理者に通知しても対処の仕様がないですし。API側でログとして出力。もしくは通知をslackに投げるとかすれば十分な気もします。

むしろ、新規登録/登録済み(status code:200番台)の場合に活用できそうな。以下みたいに。
```json
{
    "result":"UPDATE" or "CREATE",
    "user":{
         "name":"ユーザ名"
     }
}
```

あと自分がhttps://github.com/hr-js/sta_access_manager/issues/39#issuecomment-359447906　で記載したレスポンスボディは突貫で作成したので、変更の余地は確実にあります。

krhrtky, 2018-01-22:
@ikkyu-3 
指摘してくれた部分はたしかにそうだね。

自分が意識したのは、Vlaidationエラーみたいに何がダメだったのかをユーザーが知る必要があると思ったこと、なので今回の開発でって言うスコープで話せてなかったと思う、ぶれた。

ikkyu-3, 2018-01-22:
ケースバイケースってやつですかね。。

一旦自分の方で再定義(草案)を作ってもいいですか？

kakaka19, 2018-01-22:
@ikkyu-3 

status codeで分岐だとAPI側としてもありがたいです。`result`の値で迷いそうなので。

>一旦自分の方で再定義(草案)を作ってもいいですか？

自分がやってみてもいいですか？変更はあまり大きくならないと思うので、明日の夜にはできると思います。レビューはお願いしたいですが。

ikkyu-3, 2018-01-22:
> status codeで分岐だとAPI側としてもありがたいです。resultの値で迷いそうなので。

お互いがベストな方法がわかってよかったです！

> 自分がやってみてもいいですか？変更はあまり大きくならないと思うので、明日の夜にはできると思います。レビューはお願いしたいですが。

はい、お願いします。ありがとう。
レビューします！

kakaka19, 2018-01-23:
APIの草案できました。#43 の話もあるので、一旦ユーザIDでの登録処理などは外しました。

レスポンスの概要は以下の通りです。

|status|response|
|---|---|
|200|正常系|
|500|サーバ側での異常が発生。共通のレスポンスを返す|
|400|Bad Request。登録処理ですでに登録済みの場合のみ発生。|

想定している利用者のアクティビティは以下の通りです。

__初めての利用する場合__

3と4は一緒にできそうな感

1. ユーザ登録を行う。
1. ICカードの登録を行う。
1. 入室処理を行う
1. 退室処理を行う

__２回目以降利用する場合__

1. 入室処理を行う
1. 退室処理を行う

### 処理失敗(共通)

status: 500

```json
{
    "error":{
        "message": "サーバが正常に稼働していません。管理者にお問い合わせください。"
    }
}
```

### ユーザ登録

__I/O__

| name   | value  |
| ------ | ------ |
| API    | /register/user |
| method | POST   |

__Request__

Content-Type: application/json

```json
{
    "user":{
        "name": "土曜開放太郎"
    }
}
```

__Response__

#### 登録成功

status: 200

```json
{
    "user":{
        "name": "土曜開放太郎"
    }
}
```
#### 登録済み

status: 400

```json
{
    "user":{
        "name": "土曜開放太郎"
    },
    "error":{
        "message": "あなたは登録済みです。"
    }
}
```

### ICカード登録

__I/O__

| name   | value     |
| ------ | --------- |
| API    | /register/card |
| method | POST      |

__Request__

Content-Type: application/json

```json
{
    "id": "{ICcard'sID}",
    "user": {
        "name": "土曜開放太郎"
    }
}
```

__Response__

#### 登録成功

status: 201

```json
{
    "id": "{ICcard'sID}"
}
```

#### 登録済み

status: 400

```json
{
   "id": "{ICcard'sID}",
    "error":{
        "message": "このICカードは登録済みです。"
    }
}
```

### カードID更新

__I/O__

| name   | value     |
| ------ | --------- |
| API    | /update   |
| method | POST(PUT) |

__Request__

Content-Type: application/json

```json
{
    "id": "{ICcard'sID}"
}
```

__Response__

#### 更新成功

status: 200

```json
{
    "id": "{ICcard'sID}",
    "user": {
        "name": "土曜開放太郎"
    }
}
```

#### ユーザ未登録

status: 400

```json
{
    "id": "{ICcard'sID}",
    "error":{
        "message": "未登録のカードIDです。"
    }
}
```

### 入室処理

__I/O__

| name   | value  |
| ------ | ------ |
| API    | /entry |
| method | POST   |

__Request__

Content-Type: application/json

```json
{
    "id": "{ICcard'sID}"
}
```

__Response__

#### 入室処理成功

status: 201

```json
{
    "id": "{ICcard'sID}",
    "user": {
        "name": "土曜開放太郎",
        "isEntry": "true"(boolean)
    }
}
```

### 退室処理

__I/O__

| name   | value     |
| ------ | --------- |
| API    | /out      |
| method | POST(PUT) |

__Request__

Content-Type: application/json

```json
{
    "id": "{ICcard'sID}"
}
```

__Response__

#### 退室処理成功

status: 200

```json
{
    "id": "{ICcard'sID}",
    "user": {
        "name": "土曜開放太郎",
        "isEntry": "false"(boolean)
    }
}
```

### 集計

__I/O__

| name   | value          | repletion                       |
| ------ | -------------- | ------------------------------- |
| API    | /users/{:date} | `{:date}`には日付(yyyymmdd)が入ります |
| method | GET            |                                 |

__Response__

#### 一覧取得成功

入室時間で降順にソート。利用者が一人もいない場合は空の配列を返す。

status: 200

```json
{
    "users": [
        {
            "name": "hoge",
            "purpose": "自習",
            "isEntry": "true/false",
            "workspace": {
                "entry": "入室時刻",(long)
                "out": "退室時刻(しているなら)"(long)
            }
        },{
            ....
        }
    ]
}
```

kakaka19, 2018-01-23:
データ型に関しては`()`で明示されていない値以外はstringです。

ikkyu-3, 2018-01-23:
@kakaka19 
草案ありがとう！データ構造もシンプルですし、フロント側の処理もしやすいです。
見落としなど問題はなさそうです。

すいません１点だけ、
細かいことなんですが、**集計 / 一覧取得成功**の"isEntry"の値ってbooleanかなと思うので、修正お願いしてもいいでしょうか？

よろしくお願いします。

kakaka19, 2018-01-23:
@ikkyu-3 

>すいません１点だけ、
細かいことなんですが、集計 / 一覧取得成功の"isEntry"の値ってbooleanかなと思うので、修正お願いしてもいいでしょうか？

booleanですね。記載漏れてました...修正します。

krhrtky, 2018-01-24:
@ikkyu-3 
作成ありがとう！
概ねいいんじゃないかと。
細かいところで2つ気になったところがあったけど、決めの問題だと思ってる。

1. entry/out だと対義語になってない気がした。

    => 正直わかればいいでしょとも思ってる。

1. `porpose` が文字列でそのまま入ると集計しづらそう(集計するとはいっていない)

    => 文字列ベタ打ちだと汎用性がないかなと思った。

こんなところです。

kakaka19, 2018-01-24:
@krhrtky 

>entry/out だと対義語になってない気がした。
=> 正直わかればいいでしょとも思ってる。

調べたら`exit`だった。こっちの方が良さげすね。

>porpose が文字列でそのまま入ると集計しづらそう(集計するとはいっていない)
>=> 文字列ベタ打ちだと汎用性がないかなと思った。

集計はロジックをフロント/API側で持つわけじゃなくて、Elasticsearchにクエリ投げれば返してくれるから、そこまで難しくならないと思う。

1 = 勉強会
2 = 自学習
3 = その他

みたいに、数値で返すことも考えたけど、逆にその方が大変かなと。

あと、一応このissueでは利用者向け機能に関するI/Oに限定して考えてる。管理者向け機能にI/Oに関しては、仕様を相談しつつ決めて行った方がいいかなーと思ってる。

kakaka19, 2018-01-24:
あと俺のアカウントは@kakaka19っす！k始まり多くてうけるね。

ikkyu-3, 2018-01-24:
> 調べたらexitだった。こっちの方が良さげすね。

賛成です。意識してなかった、英語力ほしい。。

**porposeの種類**は
- 勉強会
- 自習
- 仕事
- 遊び

以上の４つで、モック作成をしています。

kakaka19, 2018-01-27:
@ikkyu-3 
@krhrtky 

おふた方の指摘を反映して、wikiにアップしました!

https://github.com/hr-js/sta_access_manager/wiki/api

kakaka19, 2018-01-27:
wikiにあげたのでクローズ

closed 2018-01-27 by kakaka19