hr-js/sta_access_manager #40: APIのパラメータValidation時の挙動について

krhrtky, 2018-01-21:
## 考慮すべき点

- Validationチェックにかからない
エラー(サーバ側での処理の失敗)

    statusは200、返却値resultで処理の成否を判定する。

- Validationチェックによるエラー(必須のパラメータが漏れている、方の相違)

    statusは400(bad request)で返す。

となりそうですが、400時のエラーハンドリングはフロント側で考慮可能でしょうか？
難しい場合は、自前でValidation処理を用意します。

ikkyu-3, 2018-01-21:
> 400時のエラーハンドリングはフロント側で考慮可能でしょうか？

可能という認識です。通信時は、try-catchで処理すれば拾えるかと思います。

krhrtky, 2018-01-21:
確認ありがとうございます！
ひとつ気になったのは、Ajaxは何を使って実装しますか(fetch, axios等)

fetch使う場合には、4xx,5xxでもcatchされなさそうです。
http://blog.mudatobunka.org/entry/2016/04/26/092518

ikkyu-3, 2018-01-21:
情報ありがとう！

[SuperAgent](https://github.com/visionmedia/superagent)をつかう予定でいます。
fetchではなかったと思いますが、少し考慮が必要ですかね。

krhrtky, 2018-01-22:
確認ありがとうございます。
SuperAgentのリンクをみたところ
```javascript
request
  .get('/some-url')
  .query({ action: 'edit', city: 'London' }) // query string
  .use(prefix) // Prefixes *only* this request
  .use(nocache) // Prevents caching of *only* this request
  .end((err, res) => {
    // Do something
  });
```
となっていて、`end` メソッドが処理になるかと思われます。
これを見る限り `err` でハンドリングしていそうですね。

ikkyu-3, 2018-01-24:
ですね！

[SuperAgent — elegant API for AJAX in Node and browsers](https://visionmedia.github.io/superagent/)

ここみると、エラーハンドリングの方法が紹介されてますね。
`err` オブジェクトにstatusがあるみたいです。

テストしてみたほうがいいと思いますが、
現状では、

**400時のエラーハンドリングはフロント側で考慮可能**

と考えていいかと思います。

kakaka19, 2018-01-27:
解決したようなのでClose

closed 2018-01-27 by kakaka19