hr-js/sta_access_manager #42: ドキュメントの運用ポリシーを定める

kakaka19, 2018-01-22:
目的
---
ドキュメントの運用ポリシーを作成し、[wiki](https://github.com/hr-js/sta_access_manager/wiki)に記載する。

背景
---
ドキュメントの運用を、githubのwikiで行うことが決定した。#35
よって以下のことを定める。
- 記載してはいけないもの。
- 仕様/設計などのレビューフロー

ゴール
---

- [ ] wikiに記載してはいけないものを明文化する
- [ ] ドキュメント更新の際のフローを明文化する
- [ ] 上記の二つに関してレビューを受け、項目の過不足を修正する
- [ ] 制定したポリシーをwikiに記載する

kakaka19, 2018-01-27:
追加

レビューが必要/共有事項がある場合は、関係者をAsigneeに追加する

kakaka19, 2018-02-21:
## wikiに記載するべきではない情報
### 個人情報
- 氏名
- 社内メールアドレス
- 社員番号

## ドキュメント更新のフロー
1. 作成するドキュメントに関してのissueを作成する
    - 自身とレビュワーをAsigneeにする。見ておいてほしい人もAsigneeに追加する。
    - 最低限そのドキュメントを作成する __目的__ はdescriptionに記載しておく
1. issueのコメントにドキュメントを書いていく(wikiはブランチが切れないため)
1. レビューを受ける/内容を修正する。
    - wikiのどこに書いたら良いか迷ったらこの時点で相談する
1. wikiに内容を反映する。

kakaka19, 2018-02-21:
@ikkyu-3 
@krhrtky 

すっかり忘れていたこのissueを今更やりました。時間あるときにレビューお願いします。

ikkyu-3, 2018-02-22:
@kakaka19 
確認しました。

> issueのコメントにドキュメントを書いていく

これいいっすね！

wikiに記載するべきではない情報・ドキュメント更新のフロー問題ないと思います。

kakaka19, 2018-02-22:
@ikkyu-3 

ブランチ切ってプルリク出せればいいんですけど。苦肉の策すね...

ikkyu-3, 2018-02-22:
@kakaka19 
とりあえず、これで運用してみて改善点あれば変えていきましょ

kakaka19, 2018-05-26:
wikiに記載したのでクローズ(忘れていた)

closed 2018-05-26 by kakaka19