hr-js/sta_access_manager #43: ユーザのマスターデータに関して

ikkyu-3, 2018-01-23:
当初は外部ファイルをと考えておりましたが、なんやかんやで自分たちで作ってしまったほうが楽そうです。

なので、現在選択肢として、
1. こちらで外部ファイルを作成する
1. 外部ファイル作成せず、DBで管理する
1. なんやかんやを解決する

があげられるかなと考えてます。

自分的には、わざわざファイルを作る必要がないと思って、2がいいなって思ってます。
意見などあれば、よろしくお願いします。

kakaka19, 2018-01-23:
@ikkyu-3 

>自分的には、わざわざファイルを作る必要がないと思って、2がいいなって思ってます。

自分たちでuseridと名前を紐づけたマスターテーブルを作成するってことでしょうか？

ikkyu-3, 2018-01-23:
@kakaka19 
そうなります。2は、マスターテーブルを作る認識であっています。

kakaka19, 2018-01-23:
@ikkyu-3 

わかりました。そうなると、DB作成の為にuseridと名前をセットで取得できるリソースが必要になりますね。（案1でもそうなりますが。）

自分としてはICカードIDと名前を紐づければ、利用者登録時に動的にDBにインサートしていけるかなと思っていたんですが、夜も更けてきましたので、明日、考えをまとめてコメントします。

kakaka19, 2018-01-24:
@ikkyu-3 
@krhrtky 

ちょっと考えをまとめてみました。

齟齬があるかもしれないので、背景も書いてみました。

## 背景

ユーザIDと個人名が紐づいたデータが他部署からもらえると思っていたが、セキュリティ的に厳しそうだし、時間もかかりそうな為、自分たちで用意したい。

焦点として __ID(一意な情報)を何にするか__ があると思います。

### 案1

ユーザIDを一意とする場合。

インデックス構成例

|ユーザID|名前|
|---|---|
|123456789|土曜開放　太郎|

```json
{
    "id": "123456789",
    "user":{
        "name": "土曜開放　太郎"
    }
}
```
__メリット__

- ユーザ登録時、ユーザID入力のみで登録ができる。
- __名前を入力させなくて済むため、入力ミス/管理者の意図しない名称(あだ名など)での登録を防げる__

__デメリット__

- __運用前にインデックスを用意しなければならない為、以下のタスクが発生する__
    - ユーザIDと名前が紐づいたリソース(社内SNS、社内システム...etc)を探索する必要がある
    - スクレイピングなどのリソースからインデックスに登録する為の作業が必要
    - __リソースを見つけきれなかった/取得できなかった場合の損失感が大きい。1からになる(やるなら期間を決める)__
- 中途/新卒入社の方々の情報の新規登録作業が都度発生する。バッジ処理などの工夫が必要

デメリットの中核には`自分たちだけではどうにもならん！`っていうのが起こりうるっていうのがあります。
ただ利用者/管理者側のメリットも結構でかい。

### 案2

ICカードIDを一意とする場合。

インデックス構成例

|カードID|名前|
|---|---|
|1a2b3b4c5d|土曜開放　太郎|

```json
{
    "id": "1a2b3b4c5d",
    "user":{
        "name": "土曜開放　太郎"
    }
}
```
__メリット__

- 利用者登録時にインデックスを更新する為、運用中の新規登録作業が発生しない。
    - 事前にインデックスを用意しておく必要がない。

__デメリット__

- __利用者に名前を入力させる為、入力ミス/管理者の意図しない名称(あだ名など)での登録を完全には防げない(サニタイズである程度は防げるが...)__
- 社内の他のシステムと連携する際、ICカードIDでは双方で一意な情報とできないため、面倒かも(まぁまだ決まってないことなのでそこまで)
- ICカードIDと名前を1リクエストに含める為、同期させる為の工夫が必要。websocketなら実現可能？

案２だとアクティビティはこんな感じになるのかなと思っています。

1. ICカード登録画面
2. 利用者がPaSoriにタッチ
3. Raspberry pi側からブラウザに通知 
4. 名前入力画面へ遷移
5. 利用者が名前を入力
6. ICカードIDと名前をマージしてリクエストを投げる

ikkyu-3, 2018-01-24:
@kakaka19 
ありがとうございます。結構難しい問題ですね。
思ったことまとめました。

### 案1
やはり**運用前にインデックス**が問題ですね。
メリットが大きく、利用しやすいってのが売りですが、デメリットの
> - リソースを見つけきれなかった/取得できなかった場合の損失感が大きい
> - 新規登録作業が都度発生する。バッジ処理などの工夫が必要

が気になりました。デメリットも大きいように思えます。

### 案2
> 利用者に名前を入力させる為、入力ミス/管理者の意図しない名称(あだ名など)での登録を完全には防げない(サニタイズである程度は防げるが...)

提案なんですが、
ここをある程度許容すれば(**利用者は意図しない入力をしないということを前提とする**)、現実性があるように感じます。理由としては、
1. 不特定多数向けのサービスではない
1. 現在の紙媒体の入力状況を考慮

一応こちら側でできる対応として、
- 入力する前に、注意喚起
- 確認画面による入力ミスの防止

など、**防ぎにくくする**ことは可能じゃないかなとという認識です。
名前の修正方法を考慮する必要がありますが、、

> ICカードIDと名前を1リクエストに含める為、同期させる為の工夫が必要。websocketなら実現可能？

同期させることは可能です。Reduxのstoreで管理すれば、実現できます。

アクティビティは
1. ICカード登録画面
1. 利用者がPaSoriにタッチ
1. Raspberry pi側からブラウザに通知
1. 名前入力画面へ遷移(注意喚起)
1. 利用者が名前を入力(**利用者的にkeybord用意**)
1. **確認画面を表示し、ボタンを押させる**
1. ICカードIDと名前をマージしてリクエストを投げる

となると思います。

現状や運用時を考えて、
**案2**に賛成です。
まだ考慮必要かと思いますが、どうでしょうか？

よろしくお願いします。

kakaka19, 2018-01-27:
@ikkyu-3 
>不特定多数向けのサービスではない
現在の紙媒体の入力状況を考慮
一応こちら側でできる対応として、
入力する前に、注意喚起
確認画面による入力ミスの防止
など、防ぎにくくすることは可能じゃないかなとという認識です。
名前の修正方法を考慮する必要がありますが、、

現時的なのは案２ですね。修正方法はユーザ(ICカード)更新機能で提供できるかと思います。

>現状や運用時を考えて、
案2に賛成です。
まだ考慮必要かと思いますが、どうでしょうか？

後から`やっぱり案1にしよう!`となっても移行はそれほど難しくないので、当座は案2で進めていくのはどうでしょう。

ikkyu-3, 2018-01-27:
それでいきましょう！！

krhrtky, 2018-01-27:
@kakaka19 @ikkyu-3 
案2で良いと思います。

kakaka19, 2018-01-27:
案2でいくのでClose

kakaka19, 2018-01-31:
メールアドレスに関して意見が出ましたので、一旦整理してみます。

まず、本issueでユーザインデックスのマッピングが以下(案2)で決定しました

__ICカードIDを一意とする__

インデックス構成例

| カードID      | 名前          |
| ---------- | ------------- |
| 1a2b3b4c5d | 土曜開放　太郎 |

```json
{
    "id": "1a2b3b4c5d",
    "user":{
        "name": "土曜開放　太郎"
    }
}
```

上記のマッピングでは以下の欠点があります。

- `name`の重複が起こった場合、利用者/管理者が画面上で識別が行えない。
    - 利用者は入室時間などから自身を特定できるが、管理者は`Aさん`と`A'さん`を判別できない。

管理者が識別できる情報がないのは結構痛いかも。

しかし、前提として`利用者が限定される`というのがありますので、そもそも重複が起こりえない可能性もあります。また、社内SNSで確認してみたのですが、現時点で同姓同名の方は存在しませんでした。

ただ、社内SNSの網羅性は怪しいですし、新卒/中途入社で後から重複が発生する可能性もあります。そもそも管理者が __社員に同姓同名は存在しない__ と把握しているかどうかも微妙なので、メールアドレスを識別子として持つのは決して無駄ではないです。

というわけで3つ目の案です
### 案3

入退室処理時はICカードIDを一意とし、利用者/管理者側への識別子としてメールアドレスを含める。

| カードID      | 名前          | メールアドレス            |
| ---------- | ------------- | ------------------ |
| 1a2b3b4c5d | 土曜開放　太郎 | ta-low@hoge.ne.jsp |

```json
{
    "id": "1a2b3b4c5d",
    "user":{
        "mail": "ta-low@hoge.ne.jsp",
        "name": "土曜開放　太郎"
    }
}
```

__名前の代わりにメールアドレスを使う__ という話もありましたが、機能として`土曜開放デー終了時に管理者に利用者一覧を通知する`というのがありますので、現行の業務フローと照らし合わせても、名前は必須な情報となります。

__メリット__

メリットに関しては案2の欠点を一部補完する感じです。

- 利用者/管理者が識別できる

__デメリット__

- 名前入力に加えてメールアドレスの入力が発生する
    - ユーザ登録/更新時のみなので、そこまで手間でもないか。

まぁ案３でもいいですかね。名前、メールアドレスの入力なんて、正直普通の登録フローですし。

ikkyu-3, 2018-01-31:
ユーザIDでも、メールアドレスの代わりができそうなのですが、そこはどうでしょう？
違いは
- 入力数はユーザIDが少ないです。
ってぐらいでしょうか。。欠点を解決するという視点でいうと、このぐらいしか、、

> まぁ案３でもいいですかね。名前、メールアドレスの入力なんて、正直普通の登録フローですし。

案3でもいいように思えます。今後のデータの活用(拡張的な意味で)とかで決めちゃいたいですね。
その辺はあまり得意な分野ではないので、意見もらえると嬉しいです。

kakaka19, 2018-01-31:
>案3でもいいように思えます。今後のデータの活用(拡張的な意味で)とかで決めちゃいたいですね。

利用者が限定的だという前提があるので、他で活用できるほど汎用性のあるデータは収集できないと思います。

ikkyu-3, 2018-01-31:
> 利用者が限定的だという前提があるので、他で活用できるほど汎用性のあるデータは収集できない

なるほど。ありがとうございます。

では、
**入力するユーザ視点で考えてみました。**

現在想定している、入力ハードは、テンキーなしのキーボードです。
ユーザIDは数字のみ、メールアドレスは半角英文字 + 「.」

と考えると、メールアドレスがいいかなって思います。
主観的な意見で恐縮なのですが、

案3でいくのはどうでしょうか？