hr-js/sta_access_manager #57: 目的の選択肢についての提案

ikkyu-3, 2018-02-03:
*** 現在の目的一覧
- 勉強会
- 仕事
- 遊び
- 自習

*** 提案内容
「遊び」  =>  「サークル」

**サークル目的で使用する可能性がある**かなって思いまして、提案しました。
影響範囲は、URLの変更とモック、サーバは影響範囲なしです。

よろしくお願いします。

Tchikawa, 2018-02-03:
いいと思います！！
[メモ]
・選択画面変更
・選択後画面変更
・人形のイラスト変更
・選択画面の設計書修正
・選択後画面の設計書修正

ikkyu-3, 2018-02-03:
サークルに変更します。

kakaka19, 2018-02-03:
サークル活動の方がいいんじゃないですが？

ikkyu-3, 2018-02-04:
> サークル活動の方がいいんじゃないですが？

他の目的の文字数との差を小さくしたいという考えが実はありまして。。。

krhrtky, 2018-02-04:
@ikkyu-3 
ふと思いましたが、

> 勉強会

勉強のほうがよくないですか？普通に勉強しに来てる人もいると思うし。

ikkyu-3, 2018-02-04:
@krhrtky 
> 勉強のほうがよくないですか？普通に勉強しに来てる人もいると思うし。

`勉強会 => 勉強` と変更すると、
`自習` と`勉強` の違いが曖昧になってしまい、選択する人が困ってしまう心配があります。

もし`勉強` に変更した方がいいなら、自習をやめて、新たに目的を追加したほうがいいかもなって思っています。

ikkyu-3, 2018-02-04:
@krhrtky 
了解！

**現在の目的一覧**で、勉強会の下に自習って書いておけばよかったと思ってます。
一番下に書くのは、ちょっと自分の配慮足りてなかったかなって。
こちらも気をつけます！

ikkyu-3, 2018-02-05:
`サークル` or `サークル活動` に関しては、

今後変更が容易に可能なので、暫定として`サークル` とさせてください。

すみませんがよろしくお願いします。

ということで、closeします。
ご意見ありがとうございました。

closed 2018-02-05 by ikkyu-3