hr-js/sta_access_manager #58: ラズパイの起動時のシェルを作成

ikkyu-3, 2018-02-03:
** 起動したらアプリケーションが使える状態にしたい
+ nodeを起動
+ curlコマンドで起動確認
+ Chromiumをフルスクリーン起動させて、アプリにアクセス

の順にシェルを作成してください。

OSは、Ubuntu: 17.04でもおkです。お任せします。

krhrtky, 2018-02-04:
@yumtana2 
## Vagrant の環境設定

### GUI(desktop)を有効にする

- 下記の記事を参考に設定

    [Vagrant + Ubuntu](https://qiita.com/mhagita/items/8f339106c3b48eb098ca)

    [Ubuntu destopのエラーの解消](https://teratail.com/questions/102552)

### 課題

- GUIで起動したがターミナルが起動しない。

    => 原因の確認中(もし、知っている人がいたら教えてほしいです)。

krhrtky, 2018-02-04:
ちなみに終了するshのあったほうがいいのでは？
と思い始めました。

↑と逆の順序でプロセスを終了しいくようなイメージで最後はshutdownまでするとか

ikkyu-3, 2018-02-04:
@krhrtky 
いいアイディア！！
アプリ上でどうやってshの起動させて終了させるかは、他のissueでやるとして、
shutdownまでいけると、管理者の人は助かると思います。

必須ではないですが、お願いします。

krhrtky, 2018-02-04:
@ikkyu-3 
issueの更新ありがとうございます。

@yumtana2 が担当が増えていい機会だとおもうので！
Italationはおって検討としたいです。

krhrtky, 2018-02-04:
## log

- [new branch](https://github.com/hr-js/sta_access_manager/tree/featuer/startup) 作成

- 俺なりのベースを作成(@yumtana2 はあえて見ないように)

## 課題

- error handling

- 各工程の実行方法

krhrtky, 2018-02-06:
## Vagrant の環境設定-続き

### 課題

- GUIで起動したがターミナルが起動しない。

    => localのターミナルで `vagrant up` -> `vagrant ssh` (culと同様の操作)で実行出来たのでsshで実行することとする。