hr-js/sta_access_manager #62: [2/4中必須]1st itelation の進捗状況確認

krhrtky, 2018-02-04:
## 目的

- 1/27にMilestoneに各自のissueを割り振った

- 各Itelationは２週間単位のMilestoneになるため、進捗を1週終了後に実施を決定

- まず、[1st Itelation](https://github.com/hr-js/sta_access_manager/milestone/3) について状況確認をしたい

## 確認項目

- 各自の進捗率(%)

- 完了目処

- タスクの消化が難しい場合のヘッジ方法(Milestoneの変更、Assigneesの変更 等)

- その他問題等

各自の状況について共有をお願いします。

Slackでのやり取り・決定事項も流れてしまうので、こちらに改めて残すようにしてください、

kakaka19, 2018-02-04:
elasticsearchとの接続モジュール実装中です。イテレーション内で完了予定。

Tchikawa, 2018-02-04:
4/13画面完了です。今日+2画面作成予定です。

ikkyu-3, 2018-02-04:
【Rasberry Pi側】wikiページ作成中です。
モックの画面が完成したら画像を追加するだけなので、1st Itelationで完了します。

tankette148, 2018-02-04:
進捗率：50%
・画面設計機能草案作成済(ラフ)
残：今週中にまとめ→レビュー
今週中にレビューまで終了予定です。

krhrtky, 2018-02-04:
回答ありがとうございます。

@kakaka19 
APIサーバ側は今Milestoneとかやってないけど、やったほうがいいだろうか？
管理画面は @Tchikawa  が担当してくれるそうなので、人数も増えたら必要かなと思った。
やるならこちらのrepositryのItelationにいれるのでもいいかな。

みなさん
オンスケで進んでいて良かったです。
このあたりから2nd Itelationを意識していきましょう。

次のスプリントのタスクの重さに問題ないでしょうか？
前もって多いなと感じていたら、早めに相談しましょう。

kakaka19, 2018-02-04:
@krhrtky 

 人数少ないから必要ないかな。あってもいいけど、3人くらいなら直接やり取りした方が早い

krhrtky, 2018-02-04:
@kakaka19  (cc: @Tchikawa )
了解！じゃあ口頭確認して、それをissueなりwikiに落とすカタチで！ 

@ikkyu-3 
1st Itelationしか見てなかったんですけど、次のタスクって各々見えているんでしょうか？

2nd Itelationが始まってから考えるのも時間もったいないかなと思ったんですが。

krhrtky, 2018-02-05:
@minako-nakanishi 
今後はどんな予定なんだろう。

例えば、Reactの実装に向けての準備とか？

minako-nakanishi, 2018-02-05:
そもそもなんで、社員ID入力画面なくなったの？

kakaka19, 2018-02-05:
@minako-nakanishi 
最初は多部署から社員ID+社員名のデータをもらおうと思っていたけど、セキュリティ的に難しくてもらえない＆もらうのに時間かかりそう

だから、ICカードID＋社員名+社内メールアドレスとかで管理することになった。ので、社員ID入力画面はなくなっちゃった。

minako-nakanishi, 2018-02-05:
@kakaka19
自分で入力するからデータもらわなくても良いんじゃないの？

kakaka19, 2018-02-05:
@minako-nakanishi 

もともと、社員IDを入力させるのは社員名と紐づいているから便利だねって話だった。でも、そもそも、社員IDと社員名が紐づいているデータをもらえるかわからなくなったから、社員IDを入力させるメリットもなくなった。

まぁここに色々な経緯が載ってるからみてみてよ。
https://github.com/hr-js/sta_access_manager/issues/43

krhrtky, 2018-02-07:
確認用につきclose

closed 2018-02-10 by krhrtky