hr-js/sta_access_manager #63: 【Rasberry Pi側】wikiページ作成

ikkyu-3, 2018-02-04:
## 作成するもの
- 画面遷移図
- 各画面設計
    - 画面名
    - モック画面のキャプチャ
    - 機能(あれば)

ikkyu-3, 2018-02-09:
@Tchikawa @tankette148 @minako-nakanishi @krhrtky @kakaka19 
Rasberry Pi側のwikiページ作成しました。確認してみてください。
何かあれば、ご意見よろしくお願いします。

今後変更など出てくるかと思いますが、随時更新していきます。

ikkyu-3, 2018-02-09:
共有が確認できたらcloseします。

ikkyu-3, 2018-02-10:
## レビュー
- ユーザ登録した後に、待ち受けに遷移せず、目的選択画面に遷移させたほうがいい
- ICカードの読み取りエラーを考慮されていない
- 入力チェックを考慮する
- email, name入力画面一緒でいいのではないか？
- 色がチカチカしそう
- 登録はキャンセルボタンがほしい
- 入力確認のダイアログがみえた方がいい
- 登録画面は1urlで対応する方がいい
- エラーのURLは`機能名/error` にする
- スキャンしてサーバが入退室を判断する
- 入退室にスイッチを表示させると変更できると思ってしまうのではないか？
- 参加者一覧に、現在いる人と今日いた人を表示できるようにしたい

Tchikawa, 2018-02-10:
たしかに、Email、name一緒でいいですね...

ikkyu-3, 2018-02-10:
- ラズパイで遷移できるモックを開発が必要

ikkyu-3, 2018-02-11:
[Raspberry Pi側](https://github.com/hr-js/sta_access_manager/wiki/Raspberry-Pi%E5%81%B4)

- 画面遷移図更新しました。

【モック】モックアプリ作成 #73
でまた変更となる可能性が高いため、画面遷移図更新までとします。

closeします。

closed 2018-02-11 by ikkyu-3