hr-js/sta_access_manager #67: Reactのキャッチアップ

ikkyu-3, 2018-02-05:
## 背景
今後の実装を考えると、Reactがわかる人が欲しい。

## 今後任せたいタスク
まずはPresentational Components作成(jsx)をお願いしたい。
他もできるならどんどんお任せします。

## このissueの目標
Reactの理解を深めて、実装できるようになってもらいたい。

## どうやってやるの？
[QUICK START](https://reactjs.org/docs/hello-world.html)を一通りやる。
+ 余力あればReactの本を読み切る。
期限内に終って余裕があれば、どんどん理解を深めてもらいたい。(FluxとかReduxとか)

## 不明点が出た場合、まず意識してもらいたいこと(これは必須ではなく、推奨)
1. 何がわからないか把握する(メモをとる)
1. 色々とググってみる
1. それでもわからない時、質問する(どこまでやった or 調べたのか、ヒアリングします)

## 最後に
日数的に無理な場合、リスケします。早めにアラートお願いします。
その場合、一緒に振り返って(何ができてて、何ができていないか)、改善していきましょう。
また、「○○までにできそう」なのかなど、見通しを教えてもらおうと思います。

よろしくお願いします。

minako-nakanishi, 2018-02-05:
勉強の進捗報告はここに書いた方が良いですか？

ikkyu-3, 2018-02-05:
Slackでお願いします！

kakaka19, 2018-03-24:
ゴール決めてやった方がいいかもです。

minako-nakanishi, 2018-03-24:
期限を決めてということですか？

kakaka19, 2018-03-24:
期限もそうだけど、例えばチュートリアル終わったとか、練習で画面作ってみたとか、

minako-nakanishi, 2018-03-24:
勉強した内容はSlackのopen _studyに載せてるよ。

kakaka19, 2018-03-24:
まぁそれはそれでやっててもらって。
このissueがずっと開きっぱだから、もし使ってないならクローズしてもいいんじゃない？

minako-nakanishi, 2018-03-24:
上記ゴールを達成したらクローズします

closed 2018-09-29 by kakaka19