hr-js/sta_access_manager #68: Vagrant /ubuntu環境設定

yumtana2, 2018-02-07:
### 環境構築
- Vagrant / ubuntuの環境設定
 環境：Vagrant , Ubuntu17.04,
- Vagrant のGUI設定

yumtana2, 2018-02-07:
https://github.com/hr-js/sta_access_manager/issues/58#issuecomment-362913999
実施中「Configuring grub-pc」の画面が出ず挫折。試したかったこともいまいち違ってて無念。

https://github.com/hr-js/sta_access_manager/issues/58#issuecomment-363508822
ちな、こっちはできてる気がします。

krhrtky, 2018-02-07:
@yumtana2 
どんな画面が出るんだろう？
TerminalとVritual Box のキャプチャあったらみたい！

yumtana2, 2018-02-08:
パスワードがわからなくてログインできない問題が発生してます。
これ確認しようとしたらなかった…！
[Vagrant box ubuntu/xenial64 の ubuntuユーザ の passwordについて](https://qiita.com/TKR/items/f27271c963de0033f7ff)

krhrtky, 2018-02-08:
@yumtana2

username/Password : vagrant でログインできます。
#58 の記事↓参照
https://qiita.com/mhagita/items/8f339106c3b48eb098ca

yumtana2, 2018-02-14:
- ubuntuログイン
↓
- ubuntu内のFFを起動
↓
- macがフリーズ
↓
- 栗原さん:ヘルプ強制再起動でおｋ・しゅがー:スペック確認しろ・田中:8GBだわ、4GB なわけｗｗ
↓
- 強制終了後スペック確認
↓
- **無事4GB で爆死**←いまここ

yumtana2, 2018-04-14:
1. curlで疎通確認する。
　成功条件:httpステータス200で返却される、かつ、ページタイトルが返却された場合。
2. try-catchで安心安全なブラウザ起動する。
　ブラウザ起動するときにtry-catch入れる。

kakaka19, 2018-09-29:
やってなさげなのでClose

closed 2018-09-29 by kakaka19