hr-js/sta_access_manager #80: [管理者側] APIの設計を行う

tankette148, 2018-02-17:
- APIの設計つめる
 - @Tchikawa に共有

kakaka19, 2018-02-21:
APIの設計そのものよりも、どんな目的で何を書くかをまずは検討してほしい

- 何をするAPIか？
- 何のために書くのか？
    - 実装の共有をしやすくして、属人化を防ぐ
    - 記録として残しておく
    - 実装予定のものをドキュメントにして、他の人にレビューしてもらうためetc...
- 図は必要か?
    - シーケンス図とかはあると実装がわかりやすい。しかし手間。
- 文章だけで良いか?

tankette148, 2018-02-22:
後ほど修正追記しますメモ
メソッド, URL, Data, JSON, status

- ログイン  
  - Req: POST  管理者ID pass  
  - Res: status 200 

- 目的一覧画面 
  - 表示
  - 更新・削除
  - 目的一覧の表示 今いる＆来たのが早い順 日付なしOR
  - Req: Get 
  - Res: 200 

- グラフ画面
  - グラフ描画 
  - Req: Get 
  - Res: 200 

- ユーザ一覧画面
  - ユーザ表示
  - 更新・削除
  - Req: Get 
  - Res: 200

tankette148, 2018-02-28:
めも ちょっともらった意見とかまとめときます
- セキュリティ周りどうするか
- ログイン別にする？
- 全部〜/adminついてるけどいらないのでは？

詳細土曜とかに詰める感じで

tankette148, 2018-03-03:
本日中に確認してもらい、問題なければwikiへ反映します

tankette148, 2018-03-03:
今後変更があった際には別issueで対応する

Tchikawa, 2018-03-03:
確認しましたのでクローズします

kakaka19, 2018-03-03:
これAPIの話じゃなくて、画面のURLの話じゃね？と思ったんすけど、気になったところを書いておきます。

### 全体

- htmlやformなど、画面部品にまで言及している部分があります。APIとしては特に必要のない情報なので、削除してください。
- URLに`/admin/xxx`とありますが、`管理者側のAPI`として区別するだけの識別子は必要ありません。利用者側、管理者側で共通して使用するAPIの場合、それぞれ`/admin/~`、`/user/~`と冗長になってしまいます。
- 更に言えば`/admin/users`とかは管理者権限を持つユーザなのか、管理されているユーザなのか、パスからは判断できません。

### APIごと
####  ログイン,ログイン画面取得
- 単純な画面遷移だと思うので、APIは必要ありません。画面側でルーティングを行ってください。
- 認証に関しては別に切り出したいので、一旦保留にしてください

#### 目的画面表示
#### ユーザ一覧画面表示
> クエリの有無で挙動を変更します。

クエリの有無で挙動を変更する理由が若干わかりません。

#### グラフ
- `graph`というpath自体、画面側の実装を意識したものになっています。レスポンスでグラフを返すわけではないですよね？画面でグラフを作成する場合に必要な情報は何か？といった感じでAPIを決めてください。

ikkyu-3, 2018-03-03:
修正が必要なので、openに戻します。

tankette148, 2018-03-03:
- URLの/admin/~
本日の話し合いで、管理側を別アプリに切り出しを検討したい案が出たため一旦相談
/admin~は現時点でのAPIサーバーとURLが競合する可能性があったためつけていたが、
指摘より/admin部分は削除で対応

 - ログイン,ログイン画面取得
>単純な画面遷移だと思うので、APIは必要ありません。画面側でルーティングを行ってください。
画面遷移は削除で対応、別issueでルーティング設計します 。
>認証に関しては保留
保留します こちらも別issue作成します。

- 目的画面
from toが省略された場合は
デフォルトの検索条件(ページの初期表示に使用)
・直近で入室記録がある日付の1日分のデータを返す
（直近のデータのある日はフロントで判断できないので、クエリを送信しない）
from-toクエリは検索時に使用予定
を想定してます。

 - ユーザ一覧画面表示
from toが省略された場合は
デフォルトの検索条件(ページの初期表示に使用)
・クエリがない場合、全ユーザデータ取得
from-toクエリは検索時に使用予定
を想定してます。

- グラフ
/graph →変更します
/total/users/purpose(グラフ)

追加予定
/total/users
ユーザそれぞれの統計情報返す

kakaka19, 2018-03-06:
かなり悩んでるんですが。ユーザの更新・削除のAPIは提供しない方がいいのではないかと思いました。

管理者側の要求にはありませんし、提供するのであれば、認証周りをかなりしっかり検討する必要があると思っています。

ikkyu-3, 2018-03-06:
> ユーザの更新・削除のAPIは提供しない方がいいのではないかと思いました。

確かに管理画面案でこちら側で追加した機能なので、要求ではないです。
管理者ならユーザの更新・削除は必要かなと思い、とりあえず入れてみたという感じです。

今後追加実装となる可能性はあると思いますので、
個人的にはAPIの設計は残しておいて、**今回の機能は実装しない** というのがいいかなと思います。

認証周りの検討もそうなのですが、
開発期間を考えると、最近管理画面の実装する機能が多いような気がしています。

@tankette148 
意見などあれば聞かせて欲しいです。

tankette148, 2018-03-07:
確認したところ「削除、更新機能自体をなくす」とのことでしたので、
ページについては一部変更した方がいいかと思いました。
こうなるとメインはデータを見ることになるので、
一覧ページの合体かグラフと一覧合体あたりしちゃっていいんじゃないかと思ってます。

kakaka19, 2018-03-07:
@ikkyu-3 
>管理者ならユーザの更新・削除は必要かなと思い、とりあえず入れてみたという感じです。

まぁCRUD全て実装する必要もないので、要求になければいいかなと思っています。  
管理者っていっても、実際DBの運用までやってくれるかと言ったら微妙な気もするので。

>開発期間を考えると、最近管理画面の実装する機能が多いような気がしています。

グラフ表示とかはHighchartsの仕様に依存するかと思いますんで、モックは端折ってもいいと思います。

kakaka19, 2018-03-07:
一応できました

### 処理失敗(共通)

status: 500

```json
{
    "error":{
        "message": "サーバが正常に稼働していません。管理者にお問い合わせください。"
    }
}
```

### 入室記録取得

__I/O__

| name   | value  |
| ------ | ------ |
| API    | /users |
| method | GET    |

__Request__

| key       | value    |
| --------- | -------- |
| date.from | yyyymmdd |
| date.to   | yyyymmdd |

__Response__

status: 200

```json
    [
        {
            "mail": "ta-low@hoge.ne.jsp",
            "name": "hoge",
            "purpose": "work or study or meetUp or circle or other",
            "isEntry": "true/false",(boolean)
            "workspace": {
                "entry": "入室時刻",(long)
                "exit": "退室時刻(しているなら)"(long)
            }
        },{
            ....
        }
    ]
```

### 直近の入室記録取得

__I/O__

| name   | value         |
| ------ | ------------- |
| API    | /users/recent |
| method | GET           |

__Request__

なし

__Response__

status: 200

```json
    [
        {
            "mail": "ta-low@hoge.ne.jsp",
            "name": "hoge",
            "purpose": "work or study or meetUp or circle or other",
            "isEntry": "true/false",(boolean)
            "workspace": {
                "entry": "入室時刻",(long)
                "exit": "退室時刻(しているなら)"(long)
            }
        },{
            ....
        }
    ]
```

### 集計(目的別)

__I/O__

| name   | value               |
| ------ | ------------------- |
| API    | /total/user/purpose |
| method | GET                 |

__Request__

| key        | value    |
| ---------- | -------- |
| sort.key   | date     |
| sort.order | desc/asc |

__Response__
status: 200

```json
[
    {
        "date": "yyyymmdd",(date)
        "purpose" :{
            "work": 19,(integer)
            "study": 20,(integer)
            "meetUp": 30,(integer)
            "circle": 40s(integer)
        }
    },
    {
        ...
    }
]
```

Tchikawa, 2018-03-09:
ありがたや...

tankette148, 2018-03-10:
@kakaka19 まとめてくださってありがとうございます、内容確認しました!

これで一旦コミットして、今後足りない等あった際には追加していく形にしたいと思います。

wiki作成後にこのissueを閉じたいと思います。

tankette148, 2018-03-10:
@kakaka19 
[API](https://github.com/hr-js/sta_access_manager/wiki/api)
ここに追記予定ですが、重複ありそうなので後ほど相談してから追記したいと思います。

tankette148, 2018-03-10:
追記完了

kakaka19, 2018-03-24:
TOC(目次)も追加すること。

tankette148, 2018-04-04:
すみません、閉じてませんでした。
目次追記完了　今後は別issueで対応します。

closed 2018-04-04 by tankette148