hr-js/sta_access_manager #85: [API] 物理サーバのOS変更検討

krhrtky, 2018-02-21:
# 概要

## 目的

- パッケージマネージャを使用可能にする。

## 経緯

- パッケージマネージャ`apt-get` / `apt` が使用できない問題が発生。

- 事象としては、リモートリポジトリが404を返すようになった。

- 原因は、OSのバージョンと推測される [Qiita参照](https://qiita.com/nyanchu/items/a8cfc5cf627d70d798bf)

- 現在、towerPCのOSは `Ubuntu17.04` を使用している。

- `Ubuntu` 17.x系の最新は、17.10

## 対応方針

- 17.x系だと更新が頻繁に発生して、何度もアップデートが必要になる懸念あり。

- LTS16.x系での運用に切り替えとしたい。

kakaka19, 2018-02-21:
これな。当時はなんも考えずに導入しちゃったけど、正直入れなおしたいす。

ただ、やるならUbuntuを直接インストールするんじゃなくて、Hyper-Visor(VMwareとか)をいれて、Ubuntu OSの仮想マシンを作った方がいいかもしれない。

そもそも、物理サーバを捨てて、AWSとかのクラウド移行するって手もある。

krhrtky, 2018-02-21:
> ただ、やるならUbuntuを直接インストールするんじゃなくて、Hyper-Visor(VMwareとか)をいれて、Ubuntu OSの仮想マシンを作った方がいいかもしれない。

これをするとどういうメリットがあるか教えて！

> そもそも、物理サーバを捨てて、AWSとかのクラウド移行するって手もある。

稼働する時間も決まっているからサービス次第では料金もほぼ掛からないかもしれないね。
ただ、その辺を検討する時間が今はあるのか、微妙なところ。
まずは動作する本番が動作する環境を整えてから、移行するステップを踏んでも良いのでは？

kakaka19, 2018-02-21:
>これをするとどういうメリットがあるか教えて！

Hyper-Visor(VMware)導入すると以下のメリットがある
- スナップショットで状態を保存できる
- ブラウザから仮想マシンを作成できる(操作もできる)。いちいちサーバをディスプレイに繋ぐ必要がない。
- サーバのリソースが結構多いので、アプリケーションごとに仮想マシンを作って運用ができる(今後の事も考えて)。

>稼働する時間も決まっているからサービス次第では料金もほぼ掛からないかもしれないね。
ただ、その辺を検討する時間が今はあるのか、微妙なところ。
まずは動作する本番が動作する環境を整えてから、移行するステップを踏んでも良いのでは？

これはそうだね。しばらく物理サーバで運用して、運用大変なんです~クラウドなら楽ですけど~みたいな話もできるかもしれない。

kakaka19, 2018-02-21:
あと、本番環境のOS換えたいってなったときに楽

krhrtky, 2018-02-22:
なるほど。
じゃあ `Hyper-Visor(VMware)導入` を軸に進めようと思う。
その辺の知見がないので、ちょっとお願いしたい。そんで教えてください。

クラウドは後のお話