hr-js/sta_access_manager #86: 【Raspberry Pi】モックのUX・UI修正

ikkyu-3, 2018-02-22:
## 背景
- 【モック】モックアプリ作成 #73を作成
- 実際に操作してみると、改善したい内容が出てくる可能性があるがissueが作成されていない

## このissueで行うこと
- 使用感や意見・アイディアをコメント欄に書いていく
- 検討し、反映するか決める

## 目的
- [ ] UXの向上
- [ ] 画面のデザインを決定させたい
- [ ] 足りないボタンがないかなど最終的な確認をしたい

## なぜやるのか
- Reactでの実装開始後は、デザインやUIの変更をしたくない

## 補足
Assigneesは必ず確認をお願いします。
Assignees以外にも何か意見などあればお願いしたいです。

Tchikawa, 2018-02-22:
入室画面にも戻るボタン必要でしたかねこれ！

Tchikawa, 2018-02-22:
あと、背景の色考えました！
https://github.com/hr-js/sta_access_manager/issues/65

krhrtky, 2018-02-23:
外野から失礼します。

ざっくりですが、モックを改めて見た時に感じたこと。

- ボタンのサイズ配置に統一感がない ex) `=>` 

   コンポーネントとして使いまわすにも、デザイン的にも統一感が欲しい。

- headerがないので、画面の統一感に差が出ている。

   ex) 目的選択画面 は左ペインに画像がないので余計に感じる。

   これもコンポーネントとして考えたときにもやりにくいんじゃないかなと感じてしまった。

tankette148, 2018-02-24:
- 「名前とメールアドレス入力してください」ページ
  - 画像が左下だけにならないように「名前とメールアドレス入力してください」は入力欄の上部分へ
  - 戻るボタンが欲しい
- ユーザー登録されてません
  - 戻るボタンほしい
- 目的は？画面
  - 遷移時にちらつく感じがする・・・？
　　(枠とかの大きさを他と合わせたほうがいい・・・？ )
- ユーザ一覧
  - ボタン小さい
  - 一覧文字大きさ小さくしていいのでは
  - 入退室はいるいないの表示だけでいいと思う

minako-nakanishi, 2018-02-24:
・ホーム画面及び処理完了画面以外の全ての画面に「戻る」ボタンを作った方が良いと思います。
・利用者登録画面へ遷移した時に、名前入力欄にフォーカスが当たっていると便利だと思います。
・ホーム画面の「今いるひと」を「利用者一覧」へ文言変更した方が良いです。
・処理完了画面が一定時間表示されていたら、ホーム画面へ自動遷移する処理はつけると思いますが、ホーム画面へ遷移するボタンもあった方が良いと思います。
→連続処理の時間短縮のため。

ikkyu-3, 2018-03-17:
@Tchikawa @krhrtky @tankette148 @minako-nakanishi 
意見を反映し、モックを修正しました。
確認お願いします。

## 確認方法
1.  `sta_access_manager/mock/ikkey-3/sta_mock_v2` に移動
1.  `yarn install`
1.  `npm start`

※色鮮やかなので、画面から離れてご覧ください。

意見などあれば、またここにお願いします。
一旦この確認が終わったら、実装に入ろうと思っています。

minako-nakanishi, 2018-03-18:
画面確認しました！
ユーザー登録画面の名前入力項目とメアド入力項目に必須エラーが欲しいです。

時間があれば、メールアドレス入力欄は日本語入力禁止とかにすると良いかと思われます。
CSSのime-modeってやつです。（タブレットでできるかは分からないのですが。。）
時間があればで大丈夫です。

・あと質問なのですが、利用者一覧の第一ソートは入退室項目で第二ソートは目的でしょうか？
個人的に入室中の人の名前が上の方に来た方が良いかと思ったので、第一ソートは入退室項目が良いかと思っています。
第二ソートは目的か、裏で入室時間のデータを持っていれば、入室時間でしょうか？
（個人的にはどちらでも良いと思ってます。）

全体的にOKだと思います。

よろしくお願いします。

krhrtky, 2018-03-18:
@ikkyu-3 
モックの再作成ありがとうございます。

2点気になるところがありました。

1. スキャン失敗時

  - 失敗の画面でそのまま再スキャンを促しているように見受けられます。

  - 再スキャン時に失敗した場合、画面が遷移していないと思われるので、

     動きがないのでもう一度、ICを読み込んだのかがわからない。

2. 目的選択等の画面

  - 左ペインではTOPになっているが、TOPなのか？

その他は大丈夫だと思います。

ikkyu-3, 2018-03-18:
@minako-nakanishi 
確認ありがとうございます。
> ユーザー登録画面の名前入力項目とメアド入力項目に必須エラーが欲しいです。

これは入力チェックが欲しいってことでいいですか？
モックなので、機能はつけていませんが、実装ではエラー機能を実装する予定です。

> 時間があれば、メールアドレス入力欄は日本語入力禁止とかにすると良いかと思われます。
CSSのime-modeってやつです。（タブレットでできるかは分からないのですが。。）
時間があればで大丈夫です。

ime-modeはサポートされてませんね。こちらは入力制御で対応しようと思います。

> 利用者一覧の第一ソートは入退室項目で第二ソートは目的でしょうか？
wikiに載っているAPIすると

`入室時間で降順にソート。利用者が一人もいない場合は空の配列を返す。`

となるので、表示は入室時間降順となります。

ikkyu-3, 2018-03-18:
@krhrtky 
確認ありがとうございます。
> 1. スキャン失敗時
失敗の画面でそのまま再スキャンを促しているように見受けられます。
再スキャン時に失敗した場合、画面が遷移していないと思われるので、
動きがないのでもう一度、ICを読み込んだのかがわからない。

ここでもスキャンできると想定してます。
- urlを追加して再スキャンが成功した動きができるようにする
- 再び失敗の場合は、ダイアログを表示する

で対応します。

> 左ペインではTOPになっているが、TOPなのか？

`入退室処理` の方がいいですね。
TOPという表記はなんか違う気がします。入退室処理に変更しようと思います。

minako-nakanishi, 2018-03-18:
@ikkyu-3 
①必須の入力チェックはつける予定だったんですね。失礼しました（汗）
②ime-modeサポートされていないんですね。入力制御で対応するで了解です！
③ソートも入室時間のソートで了解です！

krhrtky, 2018-03-19:
@ikkyu-3 
回答ありがとうございます。

> urlを追加して再スキャンが成功した動きができるようにする
> 再び失敗の場合は、ダイアログを表示する

これは再スキャン用の動作を追加するという認識であっていますか？

もし、追加するのは以下の理由であまり良くないかなと思いました。

- やることはICカードのスキャンであり、失敗前後で変わらない。

   => メソッドは共通としても同じことをやるのにURLが２つあるのは違和感がある。

- ダイアログが他に使うところがなければ、そのコンポーネントの設計、実装に時間をかけるのはもったいない。

   => あるあるですが、`Loading... ` のような画面をつけたほうが、他機能のIC読み取りでも利用できるのではないでしょうか(websocket 仕様時にできるか調べていないので名言はできないですが)？

> TOPという表記はなんか違う気がします。入退室処理に変更しようと思います。

承知です。TOPはあってもいいのかなと思っていますが、indexページだけだなと思って言ってみました。

### 余談

ime-modeの話が出ていたので、出処は↓かなと思っていますが、念の為。

https://developer.mozilla.org/ja/docs/Web/CSS/ime-mode

ikkyu-3, 2018-03-19:
@krhrtky 
> これは再スキャン用の動作を追加するという認識であっていますか？

その認識であっています。

> - やることはICカードのスキャンであり、失敗前後で変わらない。
=> メソッドは共通としても同じことをやるのにURLが２つあるのは違和感がある。

モックだとURLで遷移させて動作を表現していますが、
Reactで実装する際は、状態で表示を切り替えるので、URLが2つになることはないとの認識です。
この説明で大丈夫でしょうか？間違っていたらすんません。

> ダイアログが他に使うところがなければ、そのコンポーネントの設計、実装に時間をかけるのはもったいない。

ブラウザのダイアログで対応と考えています。
時間があればコンポーネントに差し替えたいなという認識です。

> 余談

そこです！ありがとうございます！

ikkyu-3, 2018-03-26:
mergeされたのでcloseします。

closed 2018-03-26 by ikkyu-3