hr-js/sta_access_manager #91: 【Raspberry Pi】webpack4へupgrade

ikkyu-3, 2018-03-18:
## 目的
webpack3からwebpack4へupgrade

## 動機
- 【Raspberry Pi】webpack.config.baseを作成 #89 のように複数ファイル作成する必要がない
- パフォーマンス向上によりビルド時間が早くなったそうです。
- 試してみたい

[リリース・webpack / webpack](https://github.com/webpack/webpack/releases)

krhrtky, 2018-03-18:
気になったことがあったので、質問です。

- デメリットはないのでしょうか、特に他の人の実装実績。

  - 設定方法を調べたときににv.4.x系のものがどれだけ出てくるのか。

  - 今のプロジェクトで上手くいかない部分がある。

  - この点は、実装が遅くなる懸念があるかと思ってます。

- 対象はどこまでか。

  - ラズパイと管理画面は別パッケージ。

  - 個人的には統一すべきだと考えます(実装時に知見を共有できるため)。

よろしくお願いします。

ikkyu-3, 2018-03-18:
> デメリットはないのでしょうか、特に他の人の実装実績。
- 現在自分が確認しているものとしては、プラグイン関係のみです。
- 現段階ですとpower-assertが該当するかと思いますが、jestに変更の予定です。jestに関しては、一度テストしますね
- 他の人の実装実績は考えてませんでした。もしバグが出たなど、実装が難しくなった場合なのですが、ダウングレードは容易と考えてます。modeを削除し、scriptを変更するのみの対応できます。

よってリスクは少ないかと思います。がどうでしょう？
[Commits · hr-js/sta_access_manager](https://github.com/hr-js/sta_access_manager/commit/aa9530d0f3e474ea790f33e37dd06c9005a5b4a4)
差分が確認できます。

> 対象はどこまでか。
- 件名通り「ラズパイ」のみを対象としています。
- 管理画面は現段階でどういう実装になるか(そもそもreact使うのかも)明記はされてないため、調節する必要はあります。

> 個人的には統一すべきだと考えます
その意見には基本的に賛成です。

ikkyu-3, 2018-03-30:
mergeされたので、closeします！

closed 2018-03-30 by ikkyu-3