hr-js/sta_access_manager #96: 【Raspberry Pi】ディレクトリ構成を検討

ikkyu-3, 2018-03-25:
## React+Reduxのディレクトリ構成を検討
ディレクトリ構成を考えてみました。
ざっと考えたディレクトリ構成です。
### 案1
見本にしたディレクトリ構成
[redux/examples/todos/src at master · reactjs/redux](https://github.com/reactjs/redux/tree/master/examples/todos/src)

```
./
    app/
        actions/
        components/
        containers/
        reducers/
        saga/
        modules/
        rootRouter.jsx  //  containerをまとめたファイル + react-router実装
        index.jsx  //  ReactDOM.renderを呼び出すファイル
    test/
        actions/
        components/
        containers/
        reducers/
        saga/
        modules/
        index.test.jsx
```
#### メリット
- 各フォルダ内がどんな役割のファイルがあるかわかりやすい
#### デメリット
- 各フォルダ内のファイル数が増えると管理がしづらい
### 案2
```
.
├── components/
│   ├── App.jsx  // containerをまとめたファイル + react-router実装
│   ├── rootReducer.jsx  // reducerをまとめたファイル
│   ├── button/
│   │   ├── button.css  // css
│   │   ├── button.jsx  // jsxファイル
│   │   ├── button.test.jsx
│   │   ├── buttonAction.js  // action
│   │   ├── buttonAction.test.js
│   │   ├── buttonContainer.js  // container
│   │   ├── buttonContainer.test.js
│   │   ├── buttonReducer.js  // reducer
│   │   └── buttonReducer.test.js
│   ├── exit/
│   ├── layout/
│   ├── not-exist/
│   ├── participants/
│   ├── purpose/
│   ├── register/
│   └── sideber/
├── index.jsx  //  ReactDOM.renderを呼び出すファイル
├── modules/
└── saga/
```
#### メリット
- フォルダ毎にtestまで完結しているので管理がしやすい
#### デメリット
- ファイルの命名規則など、事前に決めておく必要がある

案1で実装した経験としては、ファイル数が増えてくると、管理しづらさがあります。

案2は管理がしやすそうなので、このディレクトリ構成でやってみようかなって思ってます。

kakaka19, 2018-03-26:
@ikkyu-3 

案２だとページでディレクトリ切ってるのか画面部品ごと切ってるのかちょっとわからなくないですか？

>フォルダ毎にtestまで完結しているので管理がしやすい

複数のページで再利用されているcontainerがあった場合（例えばbuttonContainer.jsとかそうじゃないですかね）、テストコードが重複して、変更の際に逆に管理が大変になりそうな気もします。

ikkyu-3, 2018-03-26:
> 案２だとページでディレクトリ切ってるのか画面部品ごと切ってるのかちょっとわからなくないですか？

buttonやiconなどの名前だと画面部品だなと直感的だと思ったんだけど。
そしたら、
```
components/
    pages/
    modules/
```
というのはどうでしょう？

> 複数のページで再利用されているcontainerがあった場合（例えばbuttonContainer.jsとかそうじゃないですかね）、テストコードが重複して、変更の際に逆に管理が大変になりそうな気もします。

あ、申し訳ない。
ちょっと書き方が悪かったので補足しておくと、画面部品はcontainer.jsは作成しない方針です。
container.jsはページごとに作成します。

↑修正しておきます

ikkyu-3, 2018-03-29:
## React+Reduxのディレクトリ構成以外のディレクトリ構成
チャットで話した内容をここに記載しておきます。

- React+Redux以外のディレクトリ構成以外が混沌としている
- こちらのディレクトリ構成も考えた方がよいのではないか

ということで、一度考えたいと思います。

ikkyu-3, 2018-04-21:
mergeしたのでcloseします。

closed 2018-04-21 by ikkyu-3