hr-js/sta_access_manager #98: [api] [test] jest でエラーが発生して実行できない

krhrtky, 2018-03-27:
## 事象

- `jest` で `require()` のグローバルメソッドが呼び出せない

    ```
    // エラーメッセージ
    TypeError: require(...) is not a function
    ``` 

- ググるも原因分からず(babelっぽい)

- `jest` のチュートリアルは実行できる( `require()` を読み込めている )ので、jest自体が原因か切り分けできず。

- とりあえず、状況報告まで

kakaka19, 2018-03-27:
requireできなかったモジュールはなんですか？

krhrtky, 2018-03-27:
fastify です

kakaka19, 2018-03-27:
エラーでた時の環境と、どんなコードを実行しようとしたら起きたのか教えてほしい。

krhrtky, 2018-03-27:
- 下記、ソースを実行。

- コマンド

`NODE_HTTP=http jest` をnpm scriptで実行

-  node -v

    v9.4.0

- server.js

```javascript
'use strict';↲
const fs = require('fs');↲
const path = require('path');↲
const getOpt = () => {↲
  switch (process.env.NODE_HTTP) {↲
  case 'http2':↲
    return {↲
      http2: true,↲
      https: {↲
        key: fs.readFileSync(path.join(__dirname, '.', 'https', 'cakey.pem')),↲
        cert: fs.readFileSync(↲
          path.join(__dirname, '.', 'https', 'cacert.pem')↲
        ),↲
        passphrase: 'xxxxx',↲
      },↲
      logger: true,↲
    };↲
  default:↲
    return {↲
      logger: true,↲
    };↲
  }↲
};↲
↲
const buildServer = () => {↲
  const fastify = require('fastify')(getOpt());↲
  fastify.register(require('./routes/users'), { logLevel: 'debug' });↲
  return fastify;↲
};↲
↲
module.exports = buildServer;↲

```

- server.test.js

```javascript
const fastify = require('./server')();

test('test', () => {
    fastify.inject(
      {
        method: 'GET',
        url: '/test',↲
      },↲
      response => {
      console.log(err);
      console.log(response.statusCode);
      }
    );
    fastify.close();
  });↲
````

情報足らなければ、指摘よろしくです。

kakaka19, 2018-03-27:
server.jsとserver.test.jsは同じディレクトリ？

kakaka19, 2018-03-27:
```js
module.exports = buildServer;
```
でexportsしてるから、
```js
const buildServer = require('./server')();
```
じゃないといけないと思う。

kakaka19, 2018-03-27:
なんかshema.jsでtypoがあったんでfixしてdevelopにpushしました

krhrtky, 2018-03-31:
やり直したけど、同じエラーメッセージでないすか？

kakaka19, 2018-03-31:
jestの方では試してないっす。developからの差分をpullした？

kakaka19, 2018-03-31:
branch切って、pushした方がいいかも。同じコードで試してみる。

krhrtky, 2018-03-31:
schema.jsをdevelopからコピーして実行したから間違いないはず

kakaka19, 2018-03-31:
まぁbranch切って一旦pushしてもらっていいすか。自分で書いてもいいけど、違ったらまた手間なんで。

kakaka19, 2018-04-05:
原因はserver.jsで同期を待たずに(awaitせずに)requireしていたためです。修正をrouter_testにコミットしたので確認してください。
差分貼っときます。https://github.com/hr-js/sam_api_server/commit/28cba5e7920be8184f8cb58e6bd30135395ca3f6

kakaka19, 2018-04-05:
@krhrtky 

原因はserver.jsで同期を待たずに(awaitせずに)requireしていたためです。修正をrouter_testにコミットしたので確認してください。
差分貼っときます。https://github.com/hr-js/sam_api_server/commit/28cba5e7920be8184f8cb58e6bd30135395ca3f6

kakaka19, 2018-04-05:
あと多分この修正をすると起動できなくなるので...適宜調整してください

krhrtky, 2018-04-06:
@kakaka19 
調査、対応ありがとう！助かりますm(_ _)m

これでテストコードはかけるはずなので、今週末にでも担当のAPIで正常系の処理は書くようにします。
あと、ESLintがJestに対応してないからプラグイン入れてみてプルリク出します。

@Tchikawa 
`router_test` ブランチのテストコードを参考に書き始めてみて。
確認する中身とかは全然書かれてないけど、そのへんは想像で。

krhrtky, 2018-04-15:
@kakaka19 @Tchikawa 

jestのエラー解消しました。
原因は、関数内でreqiureが使用できなかったようです。
routes/users.jsのproivderを読み込み処理がそうなっていました。
なので、トップレベルに上記処理を移行しました。
あとは、jestのリファレンスにそってテストコードを書いていけば大丈夫です！

closed 2018-04-15 by krhrtky