hr-js/sta_access_manager #111: 【API】ルーティングの確認

krhrtky, 2018-05-04:
## 概要
- モック(旧デザイン)のレビューをした時に入退室のURLをまとめたほうがいいのでは等
話をしたが、Wiki(api)に反映されていない。

## 目的
- 時間が立ってしまったのでAPI側の実装とあわせて、おさらいと確認をしたい

krhrtky, 2018-05-12:
- wiki「api」の「集計」を「一覧」に変更。
- 500ERRORを実装。
- UPDATEは現状不要。APIから削除。
- 入退室のAPIは1つにするとしていたが、
  - 存在確認
  - 入室処理
  - 退出処理

   の3の機能に分割する。

- 存在確認については、(users/:id/status)
- エンドポイント名の変更
   - out => exit

kakaka19, 2018-05-12:
UPDATEは登録ユーザの更新なので必要では？

krhrtky, 2018-05-12:
表現が悪かったので訂正します。
現段階で不要なので初期の実装から外したということです

ikkyu-3, 2018-05-13:
@krhrtky 
APIの確認ができたけど、その話し合いでswaggerを使用してAPIを管理していこうってなったかと思います。
ですので、

**今回のこのissueの目的は、達成したと判断してclose**して、
swagger対応のissueをたてたほうがいいかなって思うのですがどうでしょう？

このままopenの状態のままになってしまうのではと思ったので、、、

krhrtky, 2018-05-13:
@ikkyu-3 
そうですね。
こちらはcloseとして別のissueを作成します。

closed 2018-05-13 by krhrtky