hr-js/sta_access_manager #115: APIのディレクトリ構成検討

kakaka19, 2018-05-26:
### 目的

API側のディレクトリ構成を検討しないまま進めてしまったため、ソースファイルの配置が機能に即しないものとなっている。実装/機能追加が進むと混乱の元になるため、再構成する

### 対象リポジトリ
https://github.com/hr-js/sam_api_server

kakaka19, 2018-05-26:
現状
```
├── LICENSE
├── README.md
├── assetes
│   └── schema.js
├── package.json
├── routes
│   ├── provider.js
│   ├── query.js
│   └── users.js
├── server.js
├── test
│   └── ...
└── yarn.lock
```

kakaka19, 2018-05-26:
- dao関係のファイル(provider.js,query.js)を `/routes`から`/infrastructure(新規で作成)`に移動
- `/assetes`がそもそもスペルミス(おそらくassets)、また、命名が機能に即していないため、`/schema`にリネーム。
- schema.jsが膨らんで来ているので、request.js、response.js、error.jsに分ける。

kakaka19, 2018-05-26:
@krhrtky 
検討案出しました。意見お願いします

krhrtky, 2018-05-26:
@kakaka19 
個人的にはルートの配下に機能ごとのディレクトリを切ると見づらいのかなとおもいました。
javaだったりrailsのように src または app を作ってその中に各ディレクトリを作ったほうが見やすいかなと。
意図的にその構成にしなかった理由があれば聞いてみたいです。

あと、 `infrastructure` は配下の中身が想像しにくいかなとも思いました。

一応、自分の素案を載せておきます。
```
├── __test__
│   ├── controller
│   └── repository
└── app
    ├── commons
    │   ├── enums
    │   └── schemas
    ├── controllers
    └── repositories
```

kakaka19, 2018-05-26:
@krhrtky 
>javaだったりrailsのように src または app を作ってその中に各ディレクトリを作ったほうが見やすいかなと。
意図的にその構成にしなかった理由があれば聞いてみたいです。

そこまでの量じゃないかなーと思って切ってなかったですが、直下にあると面倒かもしれませんね。賛成です。

>あと、 infrastructure は配下の中身が想像しにくいかなとも思いました。

`/repository`でも`/dao`でもよかったんですが、リクエストに関係ない起動時の疎通確認処理などが入るので、少し包括的な名前にしました。ただ、`/repository`にしてその下に`/infrastructure`切ってもいいですね。

素案についてですが、ディレクトリ名は複数形にしなくていいと思います。アプリケーション内での役割を示せればいいので。他の細かい疑問は下に書いています。
```
├── __test__(この表記よく見ますけどアンダーバー必要なんですかね？)
│   ├── controller(ディレクトリ名はapp下と合わせた方がいいのでは？)
│   └── repository
└── app
    ├── commons
    │   ├── enums(どのようなファイルが置かれるイメージですか？)
    │   └── schemas
    ├── controllers
    └── repositories
```

kakaka19, 2018-05-26:
`/infrastructure `じゃなくて`/init`とか`/boot`とかのがわかりやすいかもしれないですね。`/repository`とは別にして

kakaka19, 2018-05-26:
@krhrtky 
まとめると、こんな感じがいいような。
```
├── test
│   ├── controller
│   ├── repository
│   └── init
└── app
    ├── common
    │   └── enum(要不要は置いといて)
    ├── controller
    │   └── schema
    ├── repository
    └── init
```

krhrtky, 2018-05-26:
> そこまでの量じゃないかなーと思って切ってなかったですが、直下にあると面倒かもしれませんね。賛成です。

ひとまずここのディレクトリは決定で。

initって何をやるものなんだっけ？

krhrtky, 2018-05-26:
それだったら、ウェブアプリを確認する機能だから、
appのディレクトの外においたほうがよくないかな？

krhrtky, 2018-05-26:
`__test__` はjest使ってるので見るかなと
`__mock__` と同じように表現を合わせているんじゃないかな。

krhrtky, 2018-05-26:
`__test__` の配下は `app` と同じディレクトリ名の認識であってます(直し漏れ)です。
`enums` の名前は変えてもいいと思っているんだけど、
purposeとか固定値を管理する場所にしたい。
`provider.js` に直にかいてあるのは見たのだけれど、それだと管理できないと思ったので。

kakaka19, 2018-05-26:
@krhrtky 
>それだったら、ウェブアプリを確認する機能だから、
appのディレクトの外においたほうがよくないかな？

一応dockerコンテナで動かすことを考えると、アプリ自体にその疎通確認の機能は必要だから`/app`下に置いた方がいいかなと思う(このアプリの機能ですっでのを示すためにも)。

>__test__ はjest使ってるので見るかなと
__mock__ と同じように表現を合わせているんじゃないかな。

アンダーバー必要なかったらない方がいいかな。cdとか設定ファイルでディレクトリ指定するときめんどい。

>__test__ の配下は app と同じディレクトリ名の認識であってます(直し漏れ)です。enums の名前は変えてもいいと思っているんだけど、purposeとか固定値を管理する場所にしたい。provider.js に直にかいてあるのは見たのだけれど、それだと管理できないと思ったので。

それならおけです。

krhrtky, 2018-05-26:
## 残課題

- あとはどの段階で変更するか

- 今のPRをどうするか

kakaka19, 2018-05-26:
>それはgitのリポジトリの中で管理しているからアプリの一部として認識できるんじゃないかなと思うけど、それとは考え方的に違うかな？

ほかのソースはapp下にいるのにそこだけルート直下ってのも変じゃない？
あとdockerコンテナで動かすときに/appにいると都合がいいってのもある。

今のPRが消化したらでいいんじゃないかな。それまでissueは開いておこう

krhrtky, 2018-06-03:
@kakaka19 
若干詰まってない状態で進めてしまった部分があったと感じた。
- `app.js` はエントリーポイントになるのでappディレクトリ直下でいいと考えています。
- initディレクトリは中身ができるときに追加。
- schemaはcommonsにするって頭になってたのでcontollersに移動します。

kakaka19, 2018-06-04:
@krhrtky 
ディレクトリ名は複数形である必要はないので、単数形にしてください

krhrtky, 2018-06-06:
@kakaka19 
特に主義はないので、変更します。
何かの構成を参考にした感じですか？

krhrtky, 2018-06-09:
@kakaka19
了解です

kakaka19, 2018-06-10:
@krhrtky 
closeしていい？

kakaka19, 2018-06-10:
mergeされたためclose

closed 2018-06-10 by kakaka19