hr-js/sta_access_manager #117: 管理側API(management.js)と利用者側API(users.js)の統合

kakaka19, 2018-06-10:
### 目的
別々のファイルになっており、一覧性が下がっているため、一つのファイルに統合する。

### ゴール
- スキーマの統合
- URLの整理(一部競合しているため)

kakaka19, 2018-06-10:
現在実施中 https://github.com/hr-js/sam_api_server/pull/17

krhrtky, 2018-06-13:
一覧性が指すものがよくわからないなと思います。

スキーマについては、
冗長な部分があったとしても、各pathに1つ用意したほうが
1対1の関係になるしでわかりやすくなるのと明示的な変数名にすればいいので煩わしくないと思います。
※共通のエラーは除く

ファイルの1ファイルにまとめると
行数が長くなって可読性が下がるとおもいます。
controllerと同じ名前にすれば、そこだけみれば済むのも利点だと思って
今の構成にしています。

kakaka19, 2018-06-13:
@krhrtky 
>一覧性が指すものがよくわからないなと思います。

一覧性についてはAPIが一つのモジュールに並んでいた方がわかりやすいかなという感じです。この辺は意見が欲しいです。

>スキーマについては、
冗長な部分があったとしても、各pathに1つ用意したほうが
1対1の関係になるしでわかりやすくなるのと明示的な変数名にすればいいので煩わしくないと思います。
※共通のエラーは除く

重複を無視して、無理に1対1の関係にするとメンテがしづらくなります。あと明示的な変数名という言葉がちょっとよくわからないです。変数はなんにしろ何かを明示しているものだと思うのですが...。パスと同じ変数名という意味ですか？

>ファイルの1ファイルにまとめると
行数が長くなって可読性が下がるとおもいます。
controllerと同じ名前にすれば、そこだけみれば済むのも利点だと思って
今の構成にしています。

行数が長くなって可読性が下がるなら、前述のスキーマの重複は無くした方がいいと思います。また、controllerに関して言えば、ロジックを切り出しているので(provider.js)行数はそこまで増えていません。今後、増えたら分けたらいいですし、そのためにスキーマの重複をなくしています(前述したように重複があるとメンテ対象も増えるので)。

kakaka19, 2018-06-13:
あとなんも言わないでやっちゃったのですが、スキーマは利用者登録・更新と入退室管理で分けています。