hr-js/sta_access_manager #120: 目的別利用者集計APIのレスポンス確認

kakaka19, 2018-07-01:
[目的別利用者集計API](https://github.com/hr-js/sta_access_manager/wiki/api#%E9%9B%86%E8%A8%88%E7%9B%AE%E7%9A%84%E5%88%A5)のレスポンスの構造は以下になっているが、改めてみるとリストで返す意味があるか不明。

```json
[
    {
        "date": "yyyymmdd",(date)
        "purpose" :{
            "work": 19,(integer)
            "study": 20,(integer)
            "meetUp": 30,(integer)
            "circle": 40(integer)
        }
    },
    {
        "date": "yyyymmdd",(date)
        "purpose" :{
            "work": 19,(integer)
            "study": 20,(integer)
            "meetUp": 30,(integer)
            "circle": 40(integer)
        }
    },
    {...}
]
```

レスポンスのdateがそれぞれ異なる値なら何が入るかを確認したい。

kakaka19, 2018-07-01:
Requestのdateが指定した値までの日毎の集計を返すような感じだった気がします

ikkyu-3, 2018-07-01:
> レスポンスのdateがそれぞれ異なる値なら何が入るかを確認したい。

値として、開放日の日付が入る認識ですね。この日付は、チャートに直接表示される日付データだったと記憶してます。  
まぁ、あくまで自分の認識なので、

@tankette148 はどうですか ?

API作成するときに、なぜこのデータ構造にしたのか理由書いておく必要がありそうですね。

kakaka19, 2018-07-02:
@ikkyu-3 
あー開放日の日付な気がして来ました。でもこれだとpagingとか必要すね。

ikkyu-3, 2018-07-03:
@kakaka19 
pagingいるのかな？ちょっとその辺は自分では判断できないかな。。

@tankette148 の実装方針が把握できてないので、聞かないとですね。

kakaka19, 2018-07-03:
@ikkyu-3 
まぁ運用期間が長くなるほどレスポンスがでかくなるので。あるに越したことはないですね。

tankette148, 2018-07-03:
>リストで返す意味

すみません、ちょっとパッとは思い出せないです、
リストじゃなくてもいい気がします。

>Requestのdateが指定した値までの日毎の集計を返す
>値として、開放日の日付が入る

で問題ありません。
開放日ごとの理由集計用です。

>paging

開放日の日数的にページングが必要なほどか？と言われたため
データが増えた際に追加実装すればいいんじゃないか、と言う話になりました。

kakaka19, 2018-07-03:
> 開放日の日数的にページングが必要なほどか？と言われたため
データが増えた際に追加実装すればいいんじゃないか、と言う話になりました。

すげぇ言った気がする。API側の実装自体は簡単なので実装しときます。クライアント側はまぁどっちでも。

ikkyu-3, 2018-07-03:
@kakaka19 
このissueの目的は達成できた感じです？

kakaka19, 2018-07-03:
@ikkyu-3 
そっすね。closeします

closed 2018-07-03 by kakaka19