hr-js/sta_access_manager #122: [運用に関して] ラズパイ側のアプリケーション

ikkyu-3, 2018-07-03:
今後のラズパイ側の開発方針です。管理者・サーバ側は別です。

## なぜissueをたてたのか？
- 自分のコミットのペースを上げたい  
  - 今後細かくプルリク投げる予定です。レビューに時間がかかってしまった場合、開発スピードを落としたくないのが本音。
- ラズパイ側の開発ペースを上げたい。

## なぜペースをあげるのか
ラズパイ側の開発を終わらせたい。

## どういう運用にするか
### pullリクエストのレビューは任意にする
#### reviewerの負担を減らす
=> 指定するのはやめよう。やりたい人がやる。知らない技術のことでレビュー頼まれると、つらい。
#### いつマージするの？
=> マイルストーンで管理する。レビューがない場合・現状いい解決策が浮かばない場合は、マイルストーンの期間が過ぎたらマージします。

不具合とか意見あれば、その後issueたてる方針にしたいです。

### いつから
- 意見がない場合や、決めの問題となった場合、今週からこの運用でやってみたいです。

一旦動くまでは、自分である程度実装してしまいたいです。

krhrtky, 2018-07-03:
@ikkyu-3 
開発スピードを上げることは賛成です。
じゃないと他にやりたいことがあってもなかなかできないので。。。

フロントの開発の流れとして特段、課題になりそうな部分はないかなと思います。
API側でもある程度スピードアップする必要があるかと思います。

あと細かい話になると思いますが、運用だとリリース後の運用を連想してしまうかなと。
開発方針でいいかと思います。

kakaka19, 2018-07-03:
@ikkyu-3 
ラズパイ側の実装が https://github.com/hr-js/sta_access_manager/pull/121 以外にないのでなんとも言えませんが、

>reviewerの負担を減らす
=> 指定するのはやめよう。やりたい人がやる。知らない技術のことでレビュー頼まれると、つらい。

レビューは実装の共有の意味合いもありますが、フロントサイドの実装者が今んとこ @ikkyu-3 だけなので、別にいいかなとも思います。

> いつマージするの？
=> マイルストーンで管理する。レビューがない場合・現状いい解決策が浮かばない場合は、マイルストーンの期間が過ぎたらマージします。

やめた方がいいかと、slackでレビュワーを急かすぐらいでいいと思います。プルリク作成後にレビュワーに設定されると通知飛ばないので気づかなかったりします。

ikkyu-3, 2018-07-03:
@krhrtky 
> 開発方針でいいかと思います。

返信ありがとう。
そうだね。開発の方針としたいです。

@kakaka19 
> やめた方がいいかと、slackでレビュワーを急かすぐらいでいいと思います。プルリク作成後にレビュワーに設定されるとレビュワーに通知飛ばないので気づかなかったりします。

返信ありがとう。

正直、今後プルリク増えると考えると、この「急かす回数」が増えると考えてしまって、

こっちの作業の負担となるのが、ストレスに感じるんですよね。じゃぁ開発するなよって感じですが。。  

んー、

一回なしでやってみませんか？

やらされるより自主性あったほうが、いいと思うんですよね。

kakaka19, 2018-07-03:
@ikkyu-3 
> 一回なしでやってみませんか？
まぁ今の開発スピード考えるとありかもしんないですね。
今んとこ立ってる実装issueのプルリクを今後あげてく感じならそこまでコードも多くならないし。

ikkyu-3, 2018-07-03:
では一旦やってみて、判断するということにします。  
試用期限は、7月14日(土)までとして、 <= 暫定です
どう変化あったかまとめたいと思います。

ご意見ありがとうございます。

振り返りは、新しいissueたてますね。

closeしませんが、7/7日あたりでcloseしようと思います。

ikkyu-3, 2018-07-08:
closeします。

closed 2018-07-08 by ikkyu-3