hr-js/sta_access_manager #125: [振り返り] ラズパイ側の改・開発方針で開発をしてみて

ikkyu-3, 2018-07-03:
## 関連issue
[[開発方針に関して] ラズパイ側のアプリケーション](https://github.com/hr-js/sta_access_manager/issues/122)

## 振り返り内容
- 開発してみてどうだったか
- よかったこと
- わるかったこと

krhrtky, 2018-07-07:
- 開発してみてどうだったか
  - 新しいFWとかサーバーサイドのNode.jsの理解が進んだ。
  - GitHubをちゃんと？した使い方をした。
- よかったこと
  - いろいろ話し合いながら進めたので、色んな人の意見や考え方がわかった。
  - スクラッチするの大変っていうのがわかった。
- わるかったこと
  - 設計とかに時間を割いたので、なかなか実装に移れなかった。
  - 業務 + αでやろうとするとスピードに個人差が出てしまった。
  - 普通の業務と同じサイクルで回そうとしてツラかった。

kakaka19, 2018-07-07:
#122 の方針に関して。

開発ペースは上がるが、複数の実装が平行に行われた時が難しそう。このままやるなら、とりあえず @ikkyu-3 がプロトタイプ的なものをバリバリに作って、そのあとにバグフィックスなり、実装追加なりをやっていく方がいいと思いました。

レビューは必要だと思う。レビュワーからの指摘よりも、実装の共有という点で重要。あとでコード見たらわかるのはそのコードを書いた人のみ。

あと、コミットが大きくなりがちな気がしたので(一人だと意識しにくい)、その辺はメンバー同士で注意しあった方がいいと思いました。

ikkyu-3, 2018-07-08:
コメントありがとう！
@krhrtky  
> スクラッチするの大変っていうのがわかった。

我々の業務内容的に経験する機会？がなかなかないように思えるので、いい経験になった気がしますね。

> 設計とかに時間を割いたので、なかなか実装に移れなかった。

これがスクラッチのデメリット(リスク)のような気がしています。  
どのタイミングだったり、どこまでやれば実装いけるのかの判断は、実装後にまた考えてみたいですね。

現段階だと、[adobe xd](https://www.adobe.com/jp/products/xd.html)使えばもっとはやめにできのでは？とか、
またいろんな内容で振り返る時間を作りたいですね。

@kakaka19 
> 開発ペースは上がるが、複数の実装が平行に行われた時が難しそう。このままやるなら、とりあえず @ikkyu-3 がプロトタイプ的なものをバリバリに作って、そのあとにバグフィックスなり、実装追加なりをやっていく方がいいと思いました。

ですね。バリバリ作っていくのは賛成です。レビューの負担がでかくなるかもしれないけど、よろしくお願いします。

> レビューは必要だと思う。レビュワーからの指摘よりも、実装の共有という点で重要。あとでコード見たらわかるのはそのコードを書いた人のみ。

実装の共有をわかりやすくする点で、コメント(今回レビューで意図が読めない実装に対する)も意識してみようと思います。

> あと、コミットが大きくなりがちな気がしたので(一人だと意識しにくい)、その辺はメンバー同士で注意しあった方がいいと思いました。

その点は注意しあうしかないですね。
かえって小さすぎると、どうしてこの実装したのか確認する手間も増えるので、やりながらメンバーにとってのいい感じのバランスを把握していく感じになっていく気がします。

来週までopenにしておくので、また気が付いた点あればコメントよろしくです。

ikkyu-3, 2018-07-08:
1週間ためしてみて・・・

## 開発してみてどうだったか
- 開発スピードあがるのでモチベーション上がった

## よかったこと
- 実装者側としてリズム？が出てきて、やる気がでる。
- プルリク出した後、次の開発ができるのは、気持ち的によい。

## わるかったこと
- レビューは任意なので、レビューする人が固定化
  - 開発ペース優先なので、想定していることのため対策の優先順位は低い認識

ikkyu-3, 2018-07-14:
2週間やってみて、実装しやすかったので

**この開発方針で今後もやります。**

違うやり方したい場合は、提案してください。  
開発スピード優先なので、よりスピードがあがる方法あればwelcomeです。

closed 2018-07-14 by ikkyu-3