hr-js/sta_access_manager #174: エラー発生時のlog運用no

kakaka19, 2018-09-22:


kakaka19, 2018-09-22:
fastifyのloggerはファイル出力だけにしておいて、slackは別に投げる方法もできる。

https://github.com/slackapi/node-slack-sdk

krhrtky, 2018-09-22:
それで問題ないと思う。
あとは、↓の課題がクリアできれば問題ないはず。
## 課題
  - loggerのファイル出力

kakaka19, 2018-09-22:
fastifyのloggerはpinoってやつらしいから多分いける気がする

https://github.com/pinojs/pino

```
pino.destination([target]) => SonicBoom
Create a Pino Destination instance: a stream-like object with significantly more throughput (over 30%) than a standard Node.js stream.

const pino = require('pino')
const logger = pino(pino.destination('./my-file'))
const logger2 = pino(pino.destination())
```

kakaka19, 2018-09-22:
こんな感じでいけるっぽい

```js
const pino = require('pino');
const logger = pino({ prettyPrint: true }, pino.destination('./my-file'));
logger.info(new Error('error message'));
```
log

```log
[1537599863330] [32mINFO[39m (13301 on hostname): [36merror message[39m
    Error: it's error message
        at Object.<anonymous> (/Users/kojak/workspace/pino-example/index.js:3:13)
        at Module._compile (internal/modules/cjs/loader.js:654:30)
        at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
        at Module.load (internal/modules/cjs/loader.js:566:32)
        at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
        at Function.Module._load (internal/modules/cjs/loader.js:498:3)
        at Function.Module.runMain (internal/modules/cjs/loader.js:695:10)
        at startup (internal/bootstrap/node.js:201:19)
        at bootstrapNodeJSCore (internal/bootstrap/node.js:516:3)
```

krhrtky, 2018-09-22:
なら問題はないね